daynix/UsbDk #2: Uninstalling and reinstalling usbdk causes random BSODs

matlo, 2015-06-07:
Uninstalling and reinstalling USBDK  causes random BSODs:
[image]
I wasn't able to get a kernel memory dump: the MEMORY.DMP file is not written to the disk (I still think I configured it right). I tried to find a method to reproduce the problem, and I ended up running the following bash script in a msys2 shell:
```
#!/bin/bash

export MSYS2_ARG_CONV_EXCL="*"
while :
do
  msiexec /x {DE78E7B1-43E5-4278-9678-174A6EC4444E} /qn
  msiexec /i UsbDk_1.0.4_x64.msi /qn
done
```
It generates a BSOD in about 5 minutes. Opening the device and printers panel may speed up the crash. I have many devices (more than 10) connected to my computer.

dmitryfleytman, 2015-06-09:
Thanks for the report, we will investigate.
In case you get a crash dump on your system - please share with us.

dmitryfleytman, 2015-06-22:
What is you Windows version?

dmitryfleytman, 2015-06-22:
I tried on Windows 7 and 8 and didn't succeed to reproduce. Probably an interoperability issue with some specific device driver.

Could you please try to obtain kernel memory dump?

Thanks,
Dmitry

matlo, 2015-06-23:
I'm using Windows 7 x64. I'll try again to get a kernel memory dump.

matlo, 2015-06-23:
Unfortunately I wasn't able to get a kernel memory dump. I think I configured it right, the BSOD even tells the memory is dumped to the disk, but I can't find any memory dump after rebooting.
However I observed that when the USB 3.0 host controller (ASMedia XHCI Controller 1042) is disabled, there's no BSOD...

matlo, 2015-06-24:
I forgot to say the following points:
* I unplugged all devices from all USB busses before launching the crash test, so I'm pretty sure that the issue is due to my USB 3.0 host controller.
* I tried on another computer running Windows 7 x86, and the issue did not happen.

dmitryfleytman, 2015-06-24:
There are a few factors that may prevent Windows from saving crash dumps or make Windows delete crash dumps on reboot.

Please check those links:
http://www.resplendence.com/whocrashed_dumpnotwritten
http://blogs.msdn.com/b/wer/archive/2009/02/09/kernel-dump-storage-and-clean-up-behavior-in-windows-7.aspx

There is no real reason for Windows to fail save crash dumps on Usb stack crashes...

YanVugenfirer, 2015-06-24:
Hi,

Please check pages 30-34:
http://www.slideshare.net/YanVugenfirer/windows-guestdebugging-kvmforum2012

Best regards,
Yan.

matlo, 2015-06-24:
This is my crash dump configuration:
[image]
[image]
And my pagefile configuration:
[image]
(my RAM is 8192MB)

matlo, 2015-06-24:
I forgot to say that I have 44GB of free space on my system drive.

dmitryfleytman, 2015-06-25:
This is strange.

When you reboot your system after the crash and login to Windows, does is show a window with notification about system restart due to unrecoverable problem and proposition to send problem report to Microsoft?

Could you check whether crash dump present in c:\Windows\ in case you DO NOT CLOSE that Window?
Sometimes Windows deletes the crash dump when this window is closed.

In case there is no crash dump in any case, could you please check whether there are minidumps? Please post those if any.

Thanks,
Dmitry

matlo, 2015-06-25:
Hi Dmitry,
I don't get this window after rebooting. I'll see if I can get a minidump.
This morning I wasn't able to get the BSOD, that's strange.
I'm thinking about reinstalling Windows...

dmitryfleytman, 2015-06-29:
Please also check whether there are driver updates for ASMedia XHCI Controller, I see there are a few BSOD reports related to their driver.

dmitryfleytman, 2015-09-25:
Hi @matlo, do you still see this problem?

closed 2015-09-25 by dmitryfleytman

matlo, 2015-09-25:
Hi,
I reinstalled Windows and updated the drivers, and I don't have this problem anymore.