daynix/UsbDk #3: Windows cannot boot after installing UsbDk

matlo, 2015-07-23:
This was reported by several users:
* https://forum.gimx.fr/viewtopic.php?f=5&t=1246
* https://github.com/matlo/GIMX/issues/324#issuecomment-120400811

seidzade, 2015-07-23:
Hi

Short research regarding this issue ... I have installed usbdk 1_0_4 x64 separately , then up in safe mode . Moved usbdk.sys from \windows\system32\drivers to desktop, then rebooted .. As result all usb devices do not work (also in safe mode) , but windows is up , both safe and normal modes .. 

Also logged the windows startup .. See the log : 

<script src="https://gist.github.com/seidzade/12d97852d66802760224.js"></script>

seidzade, 2015-07-23:
Seems that I found the problem .. At least for my case ... Noticed that in safe mode there is some problematic device in Device Manager , InfraRed Remote Eye for windows MCE Remote Controller :

[image]

After disconnecting this device and rebooting in normal mode - everything works :) .. Exclude my MCE remote :)

seidzade, 2015-07-24:
Update : for now I tried to connect IR Eye back and reboot the system and its works .. Seems that problem is disappeared ..

dmitryfleytman, 2015-07-26:
Could it be that the issue reproduces on first boot after UsbDk installation only?

Could you try the following sequence:
1. Uninstall UsbDk;
2. Reboot the system;
3. Connect the MCE Remote;
4. Install UsbDK;
5. Reboot the system.

Does your system boot successfully after that?

dmitryfleytman, 2015-09-25:
Hi, @seidzade, do you still see this problem? Please re-open this issue if you do.

closed 2015-09-25 by dmitryfleytman

seidzade, 2015-09-25:
Hi Dmitry, no, the problem is disappeared , everything works well :) .. Tnx