daynix/UsbDk #5: Receiving -1 using libusb-usbdk for any control transfer

jgh-, 2015-08-19:
Hi

This seems to be a really interesting and promising library, so thanks for it.  I have a question though, I am trying to send control transfers to a device via libusb-usbdk (porting code that uses libusb on another OS and works):

1. List devices.  I am able to iterate through the devices and find the one I am looking for.
2. Open the device.  I am able to successfully open the device.  The redirector appears to work.
3. Get a handle for the device from libusb. This appears to work.
4. Send a control transfer.  This appears to fail with -1.  Stepping through the code it seems like it successfully sends the transfer (at least I get `TransferSuccessAsync` from `usbdk_helper.WritePipe`) but then there's some IO error that happens later on or something...I get `LIBUSB_TRANSFER_ERROR` after waiting is completed.

I've plugged the device into a hardware USB sniffer and I don't see any transfer happening at all there.

Any thoughts on what I might be doing wrong? Am I missing a step or anything?

dmitryfleytman, 2015-09-25:
Hi,

Could you please collect a driver's binary trace as described at https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt and share it with us?

Thanks,
Dmitry

dmitryfleytman, 2015-10-29:
Fixed by commit 4a0354fdcd2d9ba342199180c16a853534c52527 at version 1.0.7

closed 2015-10-29 by dmitryfleytman