daynix/UsbDk #6: UsbDk driver installation failed

celloc, 2015-09-09:
Just compiled UsbDk and tried to install.
```
C:\Users\Administrator\Downloads\UsbDk-master\Install\x64\Win7Release>UsbDkContr
oller.exe  -i
Installing UsbDk driver
WdfCoinstaller exception: WdfPreDeviceInstallEx() failed (Failed to get error de
scription for error code: 0xd)
UsbDk driver installation failed
```

dmitryfleytman, 2015-09-10:
WDF co-installer failed to install.

What is your OS version?
Are you doing this as Administrator?
Could you try reboot and re-install?

celloc, 2015-09-10:
Os is Win 7 x64 Ultimate
Version	6.1.7601 Service Pack 1 Build 7601

Yes i am Administrator.
Reboot uninstall / install still fails

If it helps you this is my hardware
Motherboard: MSI x99a sli plus
CPU: i7-5820K
GPU: Radeon R9 290

dmitryfleytman, 2015-09-10:
I see.
Does installation of official binaries succeed on your system?

celloc, 2015-09-10:
No success either.
I tried v1.03, 1.04, 1.06   Pc freezes every time during install. Closing the installer fails, shutting down windows fails, the only way to shut down is pulling the plug :/
Booting will fail afterwards. Only fix for me is returning to system restore point

dmitryfleytman, 2015-09-10:
Do you have any USB devices connected?

celloc, 2015-09-10:
Only mouse and keyboard

dmitryfleytman, 2015-09-10:
So, official build freezes for you and the one you create yourself throws an error on installation?

celloc, 2015-09-10:
Just tried installing with mouse/kb on front port usb. Still no success
Yep self compiled throws error and pre comp. completely kills the system

dmitryfleytman, 2015-09-10:
Could you please run UsbView on your system and post full USB device information?

UsbView: https://msdn.microsoft.com/en-us/library/windows/hardware/ff560019(v=vs.85).aspx

celloc, 2015-09-10:
sure thing one moment

celloc, 2015-09-10:
Ok done
I hope you get what you need

http://pastebin.com/xT920XBm

dmitryfleytman, 2015-09-25:
Ok, I see that you're running with VIA USB controller which was not tested with UsbDk yet.

Let's try to collect some more information in order to understand what's going on.

First of all, regarding you own builds - could it be driver signing problem? Please try to compile Debug configuration and enable test signing on the target system before running it.
Please see https://github.com/daynix/UsbDk/blob/master/Documentation/DriverSigning.txt for information on how to enable test signing.

Regarding the official build that freezes - could you please collect a driver's binary trace as described at https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt and share it with us?
Also could you please check whether this is a system freeze of just USB devices do not work.
In the latter case things should change on the screen (clock, for example).

Thanks

dmitryfleytman, 2015-10-29:
Please re-test with version 1.0.7 and feel free to re-open this issue if the problem still exists.
In that case please collect driver's binary traces as described at https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt.

closed 2015-10-29 by dmitryfleytman

Adrielk00, 2023-10-19:
> Ok, I see that you're running with VIA USB controller which was not tested with UsbDk yet.
>
> Let's try to collect some more information in order to understand what's going on.
>
> First of all, [...]

This works for me! Disconnected all my VIA hardware. And now work... Thank you so much.