daynix/UsbDk #7: libusb_transfer fails with status LIBUSB_TRANSFER_ERROR

matlo, 2015-10-05:
Hello,
Two GIMX users get this error when using USBDK 1.0.6 x86_64 under Windows 8.1 x64:
http://gimx.fr/forum/viewtopic.php?p=7398#p7398
http://gimx.fr/forum/viewtopic.php?p=7412#p7412
The issue does not seem to be due to the device as it happens with a Dualshock 4 and an Xbox One controller. I'm not having this issue myself with these devices, and the only noticeable difference is that I run Windows 7 x64.
It works fine for Luk when the device is plugged to a USB 2.0 port, so maybe this is related to USB 3.0:
http://gimx.fr/forum/viewtopic.php?p=7424#p7424

dmitryfleytman, 2015-10-06:
Hi,

Thanks for the report.

Could you please ask users who observe this problem to install the latest UsbDk and collect binary traces as described here: https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt

Thanks,
Dmitry

matlo, 2015-10-06:
Hi,
It is said that the PDB is not required, however running UsbDkTraceToFile.bat prints the following error:
error 0x2 loading C:\Trace\UsbDk.pdb.
traceview : info BNP0000: WPPFMT No Trace entries found in C:\Trace\UsbDk.pdb
BinplaceWppFmt Failed with status 6

dmitryfleytman, 2015-10-06:
PDB is not required indeed. I'll check what's wrong with the script.

I think it should produce binary trace log despite the errors. Could you check this?

If not could you try to give it some UsbDk.pdb?

matlo, 2015-10-06:
It doesn't produce any UsbDkTrace.etl file. I tried removing the "-pdb UsbDk.pdb" part, and it produced the UsbDkTrace.etl file. However I think it doesn't contains anything valuable. With an hex editor I can see what looks like a short header, and the rest is 0xff bytes.

dmitryfleytman, 2015-10-06:
Please share the file, let's see if it contains something valuable.

matlo, 2015-10-06:
This is the file generated on my desktop: http://gimx.fr/download/c20dd120069b784ce2401ef83e0d236d0703fc95/UsbDkTrace.etl
As I'm not reproducing the issue on my desktop it won't contain anything valuable.

I want to be sure that the instructions are correct before pushing them to the users.

matlo, 2015-10-07:
I didn't manage to reproduce the issue on Windows 8.1 x64.

dmitryfleytman, 2015-10-07:
Hi @matlo,

I looked at your .etl file and it is correct but doesn't contain any log.
It is a problem indeed, let me check this...

For now, please use PDB from https://www.dropbox.com/s/5lgao4m5ktquiha/UsbDk.pdb?dl=0
to generate traces (this is UsbDk 1.0.6 PDB for Windows 8.1 x64).

Dmitry.

matlo, 2015-10-07:
Hi,
I tested with the PDB file and I got a bigger UsbDkTrace.etl file (I replaced the file linked above).
I had a quick look at the content, there are still a lot of 0xff, but I've seen the VID/PID of the device used by GIMX.
Let me know if the file looks good to you, and I'll push the instructions to the users.
Regards,
Matlo

dmitryfleytman, 2015-10-07:
I downloaded the new file by the same link and the log is following:

[1]09A8.072C::10/07/2015-22:45:13.972 [UsbDk]CUsbDkControlDevice::AddRedirect Success. New redirections list:
[1]09A8.072C::10/07/2015-22:45:13.972 [UsbDk]CUsbDkRedirection::Dump Redirect: DevID: USB\VID_054C&PID_05C4, InstanceID: 2
[1]0004.00A4::10/07/2015-22:45:14.225 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[1]0004.00A4::10/07/2015-22:45:14.225 [UsbDk]CRegText::Dump ID: 2
[1]0004.00A4::10/07/2015-22:45:14.225 [UsbDk]UsbDkEvtDeviceAdd Entry
[1]0004.00A4::10/07/2015-22:45:14.225 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[1]0004.00A4::10/07/2015-22:45:14.225 [UsbDk]UsbDkEvtDeviceAdd Exit STATUS_SUCCESS
[0]0004.0058::10/07/2015-22:45:28.236 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[0]0004.0058::10/07/2015-22:45:28.236 [UsbDk]CRegText::Dump ID: 2
[1]09A8.072C::10/07/2015-22:45:28.237 [UsbDk]CUsbDkControlDevice::RemoveRedirect Finished successfully. New redirections list:

So it looks totally ok, so proceed with passing the instructions.

BTW, having PDB one can fetch the text log using instructions from "Converting Driver Binary Traces to Text" of https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L66

matlo, 2015-10-08:
Hi,
Thanks, I passed the instructions to the users.

matlo, 2015-10-08:
A trace provided by a user: http://www.mediafire.com/download/irj1v106100mv7o/UsbDkTrace.zip

dmitryfleytman, 2015-10-08:
Hi @matlo,
This trace log looks exactly as yours:

[3]0004.0F14::10/08/2015-20:42:43.773 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[3]0004.0F14::10/08/2015-20:42:43.773 [UsbDk]CRegText::Dump ID: 13
[3]06B0.1648::10/08/2015-20:43:21.178 [UsbDk]CUsbDkControlDevice::AddRedirect Success. New redirections list:
[3]06B0.1648::10/08/2015-20:43:21.178 [UsbDk]CUsbDkRedirection::Dump Redirect: DevID: USB\VID_054C&PID_05C4, InstanceID: 13
[0]0004.0268::10/08/2015-20:43:21.342 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[0]0004.0268::10/08/2015-20:43:21.342 [UsbDk]CRegText::Dump ID: 13
[1]0004.0268::10/08/2015-20:43:21.342 [UsbDk]UsbDkEvtDeviceAdd Entry
[1]0004.0268::10/08/2015-20:43:21.342 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[1]0004.0268::10/08/2015-20:43:21.342 [UsbDk]UsbDkEvtDeviceAdd Exit STATUS_SUCCESS
[1]0004.0F14::10/08/2015-20:43:27.641 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[1]0004.0F14::10/08/2015-20:43:27.641 [UsbDk]CRegText::Dump ID: 13
[0]06B0.1648::10/08/2015-20:43:27.641 [UsbDk]CUsbDkControlDevice::RemoveRedirect Finished successfully. New redirections list:

There are no error indications. Could you please double check that the issue was reproduced indeed during this run?

matlo, 2015-10-09:
Hi,
Here is another trace provided by the same user: http://www.mediafire.com/download/ng1fqv7k6cpgs1p/UsbDkTrace2.zip

dmitryfleytman, 2015-10-11:
The log is the same (below). This makes me think the problem happens somewhere on application level. Will do some tests and post an update here.

[2]0004.11AC::10/09/2015-17:12:29.744 [UsbDk]CRegText::Dump ID: USB\VID_16C0&PID_0478
[2]0004.11AC::10/09/2015-17:12:29.744 [UsbDk]CRegText::Dump ID: 13
[3]0004.0EE8::10/09/2015-17:13:05.562 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[3]0004.0EE8::10/09/2015-17:13:05.562 [UsbDk]CRegText::Dump ID: 13
[2]0F6C.06E8::10/09/2015-17:14:45.132 [UsbDk]CUsbDkControlDevice::AddRedirect Success. New redirections list:
[2]0F6C.06E8::10/09/2015-17:14:45.132 [UsbDk]CUsbDkRedirection::Dump Redirect: DevID: USB\VID_054C&PID_05C4, InstanceID: 13
[0]0004.0EC4::10/09/2015-17:14:45.296 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[0]0004.0EC4::10/09/2015-17:14:45.296 [UsbDk]CRegText::Dump ID: 13
[1]0004.0EC4::10/09/2015-17:14:45.297 [UsbDk]UsbDkEvtDeviceAdd Entry
[1]0004.0EC4::10/09/2015-17:14:45.297 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[1]0004.0EC4::10/09/2015-17:14:45.297 [UsbDk]UsbDkEvtDeviceAdd Exit STATUS_SUCCESS
[1]0004.1498::10/09/2015-17:15:41.041 [UsbDk]CRegText::Dump ID: USB\VID_054C&PID_05C4
[1]0004.1498::10/09/2015-17:15:41.041 [UsbDk]CRegText::Dump ID: 13
[2]0F6C.06E8::10/09/2015-17:15:41.041 [UsbDk]CUsbDkControlDevice::RemoveRedirect Finished successfully. New redirections list:

dmitryfleytman, 2015-10-29:
Fixed by commit 4a0354fdcd2d9ba342199180c16a853534c52527 at version 1.0.7

closed 2015-10-29 by dmitryfleytman

matlo, 2015-11-01:
Hi Dmitry,
I've got positive feedback: http://gimx.fr/forum/viewtopic.php?p=7659#p7659
Thanks!