daynix/UsbDk #9: Installing USBDK may make the system unstable

matlo, 2015-10-20:
This was reported by on the GIMX forum:
https://forum.gimx.fr/viewtopic.php?f=5&t=1229
The PC is a Dell Latitude E5530 laptop, and the OS is Windows 7 64bit.
Issue first happened with USBDK 1.0.4. Unplugging the laptop from the docking station solved the issue. I think the docking station is [model XX6F0](http://accessories.us.dell.com/sna/productdetail.aspx?c=us&cs=6099&l=en&sku=331-6307), which interestingly has USB 3.0 ports, and most probably a USB 3.0 HUB.
After installing USBDK 1.0.6 the system became unstable again, and unplugging from the docking station doesn't help anymore.

dmitryfleytman, 2015-10-29:
Hi @matlo,

While I cannot reproduce this exact problem I have reasons to suspect that it was fixed by commit 4a0354fdcd2d9ba342199180c16a853534c52527 at version 1.0.7

Please ask the problem reporter to check if the issue still exists on version 1.0.7 (just released). In case it does please ask him to collect binary trace as described at https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt (the latest code base contains fixes for the binary tracing collection scripts problems observed on previous release).

Thanks,
Dmitry

matlo, 2015-10-30:
Hi,
I've asked the user to test USBDK 1.0.7. Thanks!

dmitryfleytman, 2015-11-16:
Closing this issue for now. Please re-open in case this problem still exists.

StuCZE, 2016-01-26:
Hi 
I have tried to install USBDK for GIMX but it wants to restart my pc and after that all USB components like USB 3.0 GIGABIT Network adapter or USB 2.0 Hub stops work.
I can use GIMX without this USB components but I need them for work. So I had idea. I can install USBDK before I start GIMX and than I can uninstall USBDK for work - but I cannot uninstall it! When I try to uninstall USBDK it shows just "preparing to uninstall" but nothing happen. Second problem is I cannot turn off my pc. It still shows "Turning off" and nothing happen. I must kill power for turn off pc and than start in recovery mode and restore my computer to the old compilation. 
Thank all for any idea.
Stu - (Dell Ispiron 7710 + Windows 7 pro x64)

dmitryfleytman, 2016-01-26:
Which UsbDk version are you using?

StuCZE, 2016-01-26:
I have tried v1.00-11 at last but I had same problem with version included in GIMX.

matlo, 2016-01-26:
Hi,
The latest GIMX installer provides USBDK 1.0.7.

dmitryfleytman, 2016-01-26:
Please collect binary trace for UsbDk 1.00-11 install and uninstall at described here:
https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52

StuCZE, 2016-01-26:
[UsbDkTrace.zip](https://github.com/daynix/UsbDk/files/105021/UsbDkTrace.zip)
Here you are. Let me know it is what you need (I hope so).

StuCZE, 2016-01-26:
[UsbDkTrace.zip](https://github.com/daynix/UsbDk/files/105044/UsbDkTrace.zip)
Here is second file. Sorry the first I did in USBDK Master, is it all what you need?

StuCZE, 2016-01-26:
So I found a solution for me. It isn't regular solution but it is usable.
My solution: If I tried to uninstall the USBDK library by the Controll Panel in Windows it freeze every time. I have just reed some posts and now I know that I can use the UsbDkController to install and uninstall the USBDK. I have created two .bat files to install and uninstall the USBDK and now I can do it every time if I need use the GIMX. If you find a regular solution for me let me know.
Tank you very much. :-)

dmitryfleytman, 2016-01-27:
Unfortunately the log is empty. Please make sure you did everything according to the manual and there were no errors during the process.

StuCZE, 2016-01-27:
Ups. What I did wrong? Where I have to place the TOOLS directory?

dmitryfleytman, 2016-01-27:
Doesn't matter. Just make sure there are no errors in script log.

dmitryfleytman, 2016-01-27:
And the script is running when you reproduce the problem.

StuCZE, 2016-01-27:
The UsbDK have to be instaled if I run the UsbDkTraceToFile.bat ?

dmitryfleytman, 2016-01-27:
No. You may run the script first and then install/uninstall UsbDk.

dmitryfleytman, 2016-01-27:
Also you may try to build UsbDk from master and collect the text log:

https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L21

StuCZE, 2016-01-27:
Problem was between chair and screen. :-)

StuCZE, 2016-01-27:
[Trace.zip](https://github.com/daynix/UsbDk/files/106246/Trace.zip)
There are two files. One with problematic USB HUB and one without it. But the USBDK required restart PC. Have I to restart it?

dmitryfleytman, 2016-01-27:
Yes, you have to restart. Does this happen on reboot only? What do you see in device manager after reboot? Do you see usb host controllers marked by exclamation sign? What is the error shown in device properties for the Usb host controller?

dmitryfleytman, 2016-01-27:
Ok, now log is not empty but my PDB does not match. What version is this?

StuCZE, 2016-01-27:
I don't see the devices (in device manager) immediately after I install USBDK and I see them immediately after I uninstall USBDK. I don't know the version of USBDK. It's the version included in GIMX 5.3. Should I install USBDK 1.0.11?

StuCZE, 2016-01-27:
Windows freezes in boot with USB devices plugged in after I install the USBDK and restart PC. I must unplug them for start PC.

dmitryfleytman, 2016-01-27:
Yes, please collect the same log with UsbDk 1.0.11

StuCZE, 2016-01-27:
Here is log with UsbDk 1.0.11 with restart but I cannot uninstall it now by UsbDkController. :-( 
I must restore my PC now. 
[UsbDkTrace_Install_with_restart.zip](https://github.com/daynix/UsbDk/files/106356/UsbDkTrace_Install_with_restart.zip)

dmitryfleytman, 2016-01-27:
Thanks, I'll take a look.

ybendito, 2017-07-16:
Unfortunately, the symbols of 1.0.11 were lost and we can decode the log only partially (we can see only 2 connected USB devices: BlueTooth and Camera)
If the problematic system is still available and the problem can be reproduced:
1. before installing UsbDk please use the UsbView tool ( https://drive.google.com/open?id=0B9gp3tZmgMhdR25XLVl5VnJMdlU ): run it, then use File-Save as txt and attach resulting file
2. From administartor command prompt run attached batch file to collect logging info of install of UsbDk 1.0.19 (rename it ro .BAT)
[UsbDkLogman.txt](https://github.com/daynix/UsbDk/files/1151250/UsbDkLogman.txt)
3. Try installing UsbDk 1.0.19 ( https://github.com/daynix/UsbDk/releases/tag/v1.00-19 ), let us know how it works. Stop the recording after attempt to install, zip and attach the recorded log.

Thanks in advance

dmitryfleytman, 2017-07-25:
Closing for now. Please reopen in case problem still exists.

closed 2017-07-25 by dmitryfleytman