daynix/UsbDk #10: SetConfiguration

msbroadf, 2015-11-03:
Is there a reason why SetConfiguration is missing? I notice that SetAltInterface is there but seems like SetConfiguration would be quite useful also

dmitryfleytman, 2015-11-06:
This operation is not supported right now because WDF does not support switching device configurations so this feature requires a bunch of WDM code to be written.

Multiple configuration support is in our plans but we cannot provide specific ETA for it right now.
And of course, patches are welcomed.

msbroadf, 2015-11-08:
OK looking at (Msg 8 here http://www.osronline.com/showThread.cfm?link=87757 )

seems like its not too hard, i might have a go at this.  I need this feature for the ipad as it uses configuration 4

dmitryfleytman, 2015-11-09:
Great! Let me know if you need any help.

jgh-, 2015-11-19:
an alternative that seems to work in my testing is just sending the control transfers to get and set the configuration directly:

```
// to get current config
libusb_control_transfer(handle, USBmakebmRequestType(kUSBIn, kUSBStandard, kUSBDevice), 0x08, 0, 0, &currentConfig, 1, 1000);
// to set current config to 3
libusb_control_transfer(Handle, USBmakebmRequestType(kUSBOut, kUSBStandard, kUSBDevice), 0x09, 0x3, 0, NULL, 0, 1000);
```

msbroadf, 2015-11-20:
I dont think that will work correctly because windows needs to build the internal structures that are related to the active configuration, like the interface and pipe structs so i suspect there will be issues that way. If you try to access a particular interface in the configuration of pipe i suspect you will get a bsod at some point...Actually now microsoft has release the KMDF as opensource you can see how they build the internal structures on the particular call. e.g https://github.com/Microsoft/Windows-Driver-Frameworks/blob/master/src/framework/shared/targets/usb/fxusbdevice.cpp#L1423

jgh-, 2015-11-20:
I haven't tried sending any transfers to endpoints within those configurations, so you could be correct.  FWIW changing configurations like this works on other operating systems without issue (or at least I haven't run into any issues so far) but perhaps windows is a special snowflake in this regard too.

msbroadf, 2015-11-20:
With linux/macosx/windows  usually there are separate calls to change configuration rather than using the generic control transfers, as they have OS housekeeping to do when they switch configurations like building their internal structures. So im not sure you can call the control request directly in any OS, but if its working for you...i checked libusb source and they issue the SET_CONFIG request to linux so it doesnt use the control transfer either, i think you will find issues when directly setting the config...

jgh-, 2015-11-20:
OK well I'll keep my eye out for your PR :)

jgh-, 2016-01-09:
I'm just going to bump this issue and ping @msbroadf - you are correct in that just setting the config via a control transfer doesn't work in Windows.  In the next few days I will embark on (hopefully) creating a patch for this unless someone else has one farther along lmk.

msbroadf, 2016-01-10:
Hi, yeah sorry I haven't started yet. To make this pull request will take probably a month of work for me. 
 If I do end up using usbdk in my project i will make this fix as its a must for me.

closed 2017-02-02 by dmitryfleytman