daynix/UsbDk #12: Clear Halt (stall)

msbroadf, 2015-11-10:
There is no way to clear a Stall on an Endpoint. Looking at https://github.com/dmitryfleytman/libusb/blob/usbdk-backend-v3/libusb/os/windows_usbdk.c it appears you are calling ResetPipe but i think that only toggles the host controller side of the pipe rather than clearing the actual stall condition on the device. I think there needs to be a method in UsbTarget that sends a Synchronous Control Request with the URB_FUNCTION_SYNC_CLEAR_STALL urb

dmitryfleytman, 2015-11-11:
According to https://msdn.microsoft.com/en-us/library/windows/hardware/hh968307(v=vs.85).aspx:

"The reset-pipe request clears the error condition in the device and the host controller hardware. To clear the device error, the USB driver stack sends a CLEAR_FEATURE control request to the device by using the ENDPOINT_HALT feature selector. The recipient for the request is the endpoint that is associated with the pipe."

Do you think any thing else is needed in order to clear an endpoint  STALL?

msbroadf, 2015-11-11:
Yeah im not sure then, i think i need to test with a usb protocol analyzer, according to https://msdn.microsoft.com/en-us/library/windows/hardware/ff551156(v=vs.85).aspx it only sends a URB_RESET_PIPE, but maybe lower down the stack it also sends the CLEAR_HALT...

dmitryfleytman, 2015-11-12:
Thanks! Since we don't have USB analyser here it would be great if you check this.

msbroadf, 2015-11-12:
I have a mercury t2, i will try to do this in the next week or so...

dmitryfleytman, 2015-12-01:
Hi,

Any luck with USB analyser?

Thanks,
Dmitry

msbroadf, 2015-12-01:
Hi Dimitry, sorry i havent got around to doing this yet, ill do my best to work on it next week, ive been busy with work...

msbroadf, 2015-12-10:
OK, i got around to testing this, yes it does send a CLEAR_HALT to the USB endpoint

closed 2015-12-10 by msbroadf