daynix/UsbDk #13: Installing UsbDk 1.00-7 on Windows 10 appears to cause USB ports to die temporarily

jgh-, 2015-11-19:
Hi, 

I am using a Toshiba Satellite C55-5240 with Windows 10 64-bit installed.  I installed UsbDk 1.00-7 and after the installation my USB keyboard and mouse stopped working until I rebooted the computer.

I'm afraid I can't really give you much more detail than that, but if you have logfiles located somewhere I would be happy to provide them.

dmitryfleytman, 2015-12-01:
Hello,

Thanks for your report.
If the issue reproducible? If so, could you please reproduce it and collect binary logs as described at https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52 ?

Please also try recently released version 1.0.8.

Thanks,
Dmitry

dmitryfleytman, 2015-12-28:
Please re-open and provide logs in case this issue still exists.

closed 2015-12-28 by dmitryfleytman