daynix/UsbDk #14: Support usb smartcard

nhancom3d, 2015-12-07:
(Sorry because my English is very weak)
Hi team UsbDK (@blochl n @ezzze),
I am building cloud system for create and run virtual machines on the server using KVM. I use the tool virt-viewer to connect to VM via protocols on spice. Spice-gtk library available Usb-redirection feature supports the connection of USB devices with a virtual machine, based on libusb and WinUSB backend, but it slow and unstable. So my aim is to integrate backend UsbDk (replace WinUSB) on Usb-redirection features of spice-gtk.  I have found a completed modified version of libusb, use usbdk-backend (refer https://github.com/dmitryfleytman/libusb/tree/usbdk-backend-v4). I have build it and integrate into virt-viewer replaces the original library libusb. It works well with USB storage devices and smartphones. However I have a problem with the device type smartcard USB token: After performing redirect USB token devices, although the operating system (Windows 7) in virtual machines see usb devices (as a reader) and install the driver successfully, however, the software reads the card information my notice not found smartcard. I've also tried many different software like "Gemalto SmartDiag" to take state smartcard but the software also did not detect any smartcard. I tried watching the Usb log and Windows Event Log, but could not find the cause. Can you guide me somehow to me debug this error, or you need me to provide further information to diagnose this problem?
Sincerely thanks !

Here is my environment:
Server: Centos 6.6 – Qemu 2.4.0.1 – Spice-server: 0.12.5
Client: Windows 7 – 32bit - UsbDK 1.00-8 – virt-viewer 2.0
I included the UsbDk log when I plug and perform redirect usb token device.

[smartcard.txt](https://github.com/daynix/UsbDk/files/53978/smartcard.txt)

dmitryfleytman, 2015-12-08:
Hello @nhancom3d, thanks for you report!

What USB device are you trying to redirect? Are you using the official 1.0.8 build?

According to the log it was collected from some build with mismatching PDB, some strings do not have format information:

> 00000018	19.99963760	[3244] Unknown( 32): GUID=dc9bef67-e666-5d4a-7c31-d2d3f8851030 (No Format Information found).	
> 00000019	19.99968910	[3244] 	
> 00000020	19.99976158	[3244] Unknown( 57): GUID=dc9bef67-e666-5d4a-7c31-d2d3f8851030 (No Format Information found).

Could you please run the official 1.0.8 binary (download it here from releases section) and collect the binary log as described at: https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52

Thanks,
Dmitry

syshack, 2015-12-09:
@nhancom3d  My environment is almosy the same as you ,but  redirect USB DISK failed. Can u give me some help? thx

My environment:
USBDK: http://www.spice-space.org/download/windows/usbdk/UsbDk_1.0.8_x86.msi
LIBUSB: https://github.com/dmitryfleytman/libusb/tree/usbdk-backend-v5
spice-GTK: spice-gtk-0.25
virt-viewer: 1.0
spice-server:0.12.5
qemu:qemu-kvm-rhev-0.12.1.2-2.450.sdcos.x86_64

nhancom3d, 2015-12-09:
Thank you for support, @dmitryfleytman 
I use a device called like "ePass2003Auto" (usbinfo)
I've reinstall the official 1.0.8 build, then perform logging as your guide. I perform the following steps:
- Running UsbDkTraceToFile.bat file to start logging
- Plug the USB token device, wait 30 seconds. 
- Connect to a virtual machine using virt-viewer 2.0.
- Start redirect USB token device, wait 30 seconds
- Stop redirect USB token and virt-viewer windows closed.
- Wait 30 seconds, ending logging and unplug usb device token
I repeat 3 times, here is my log. If you need anything, please tell me. Thanks.

nhancom3d, 2015-12-09:
Thank you for support, @dmitryfleytman 
I use a device called like "ePass2003Auto" (
[usb_info.html.txt](https://github.com/daynix/UsbDk/files/56370/usb_info.html.txt)
)
I've reinstall the official 1.0.8 build, then perform logging as your guide. I perform the following steps:
- Running UsbDkTraceToFile.bat file to start logging
- Plug the USB token device, wait 30 seconds. 
- Connect to a virtual machine using virt-viewer 2.0.
- Start redirect USB token device, wait 30 seconds
- Stop redirect USB token and virt-viewer windows closed.
- Wait 30 seconds, ending logging and unplug usb device token
I repeat 3 times, here is my log (
[binaryLog_usbdk1.0.8.zip.txt](https://github.com/daynix/UsbDk/files/56368/binaryLog_usbdk1.0.8.zip.txt)
). If you need anything, please tell me. Thanks.
p/s: I can't upload html and zip file. Please remane extension file to correct.

nhancom3d, 2015-12-09:
@syshack: Please describe in more detail about the errors you encounter. Can you install the "usbclerk" service? If yes you need turn off UsbClerk service.

syshack, 2015-12-09:
@nhancom3d I uninstalled the usbclerk and rebooted. when i plug one UDISK in and click the redirect checkbox, raise the error "-12" .

syshack, 2015-12-09:
@nhancom3d I see you use ePass2003Auto from "飞天诚信", are you chinese，if yes ，can i contact you with QQ or wechat.

dmitryfleytman, 2015-12-09:
Hi @syshack, thanks for your report,

Please collect the binary log as described at: https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52 and provide it to me.

I'd also suggest opening a separate issue for that to make things more clear.

Best Regards,
Dmitry

dmitryfleytman, 2015-12-09:
Hi @nhancom3d, thanks for the log, I'll investigate it and get back to you.

nhancom3d, 2015-12-09:
@dmitryfleytman: thanks for support
@syshack i'm not chinese, I does not use wechat or QQ, you can email me.
I think if you do not redirect any USB devices you are probably having problems with spice-gtk. You should upgrade to version spice-gtk 0:28 or later and check the spice-gtk log and virt-viewer first.

syshack, 2015-12-09:
@nhancom3d  @dmitryfleytman 

thx all. I opened a new issue for my problem

 https://github.com/daynix/UsbDk/issues/16

nhancom3d, 2015-12-14:
Hi @dmitryfleytman. Have a good week !
Do you have any difficulty in analyzing this problem? Do you want me to perform more orther tests? I can help you debug.

dmitryfleytman, 2015-12-14:
Hi @nhancom3d,

I didn't get to this problem yet, sorry.
I'm going to work on you report towards the end of this week.

Thanks for your patience,
Dmitry

nhancom3d, 2015-12-14:
Thanks @dmitryfleytman, waiting for good news from you.

dmitryfleytman, 2015-12-20:
Adding text log extracted from binary trace files.(https://github.com/daynix/UsbDk/files/67735/UsbDkTrace_GitHub_Issue14.txt)

dmitryfleytman, 2015-12-20:
Hi @nhancom3d,

I looked into the log but see no problem indications there. Device either stuck or guest device driver just doesn't access it for some reason. I have a few questions:

1. I understood that you upgraded to UsbDk 1.0.8, did you build libusb with new patches [1] as well?
2. What is your guest operating system?
3. Are you using the latest scripts [2] for log collection?

[1] https://github.com/dmitryfleytman/libusb/tree/usbdk-backend-v5
[2] https://github.com/daynix/UsbDk/tree/master/Tools/Trace

nhancom3d, 2015-12-21:
Thanks @dmitryfleytman, 
I use usbdk-backend-v4. I will upgrade it to usbdk-backend-v5, follow your instructions and send your detailed log.

nhancom3d, 2015-12-21:
Hi @dmitryfleytman,
I have followed your instructions.
I have upgraded usbdk-backend to v5. I also using your patches [2] and UsbDk build at #17
My guest (VM) is Windows 7 64bit. Here is my binary logs: [new_log_usbdk_1.0.8.zip](https://github.com/daynix/UsbDk/files/68085/new_log_usbdk_1.0.8.zip)
Thanks for support !

dmitryfleytman, 2015-12-28:
Hi @nhancom3d,

Thanks for the new logs. They provide some additional information but I'm missing one more step.
After you redirect the device please run some application that actually uses it and do actual actions that lead to the problem you observe.

Also would you please do the same as for issue #14:

1. I’ve just released the new UsbDk version - 1.0.10 (https://github.com/daynix/UsbDk/releases/tag/v1.00-10). It contains extended tracing facilities so I’d like to ask you to collect new logs with this version, UsbDk will trace much more info about what is happening w.r.t. your device.

2. Please also do the same experiment (and collect logs) with USB selective suspend disabled on the host:

Go to control panel-->Power Options—>Edit plan settings->Change advanced power settings, under USB settings—>USB selective suspend setting change both "On battery”/“Plugged In” to disabled and reboot the system.

3. Besides that, please collect usb tree dump from your system using UsbView (on both the host and the VM):

* See https://github.com/daynix/UsbDk/blob/master/Documentation/UsbInfo.txt

Thanks,
Dmitry

nhancom3d, 2015-12-28:
Hi @dmitryfleytman,
Thanks to the support very conscientious of you.
I've followed your instructions and perform this scenario:
- Running UsbDkTraceToFile.bat file to start logging
- Plug the USB token device, wait 30 seconds.
- Collect Usb tree dump on host
- Connect to a virtual machine using virt-viewer 2.0.
- Start redirect USB token device, wait 30 seconds
- Collect Usb tree dump on VM
- Stop redirect USB token and virt-viewer windows closed.
- Wait 30 seconds, ending logging and unplug usb device token

This is my log I've collected ([smartcard_log_20151228.zip](https://github.com/daynix/UsbDk/files/73181/smartcard_log_20151228.zip)). I hope it contains enough information as you expect.

Thanks.

dmitryfleytman, 2015-12-28:
Hello @nhancom3d,

How do you use your USB token? Does it come with some special accompanying application?

Thanks,
Dmitry

nhancom3d, 2015-12-28:
Hi @dmitryfleytman,
I use a USB token to authenticate the user with the website of my company. Usb token only contains information about the type of smartcard. It uses standard Windows drivers, do not need a special driver to work.
For test you only need install a smartcard reader information tool on VM such as Gemalto company's tool at here: (GUI and command line tools: http://support.gemalto.com/index.php?id=download_tools). In case it works SmartDiag Diagnosis Tool will display the message "Pass" like this 
[image] This is a picture I took at the physical machine - this is my result I expect.
Thanks very much.

dmitryfleytman, 2015-12-28:
Hi, @nhancom3d 

So during log collection, after you started the redirection, could you run this installation process on the guest side? I need logs produced by UsbDk at the moment when this installer tries to connect to the device and fails.

Thanks,
Dmitry

nhancom3d, 2015-12-28:
Hi, @dmitryfleytman,
I sent you log UsbDk in case I make redirect and run Gemalto SmartDiag Diagnosis Tool: [log_usbdk_1.0.8_smardcard_20151228_v2.zip](https://github.com/daynix/UsbDk/files/73235/log_usbdk_1.0.8_smardcard_20151228_v2.zip)
Hope it useful for you. Thanks.

dmitryfleytman, 2015-12-28:
@nhancom3d, are you still on version 1.0.8? If yes, could you please do the same with v1.0.10?

Thanks.

nhancom3d, 2015-12-28:
Hi, @dmitryfleytman,
No, I am not on version 1.0.8. I've upgraded to version 1.0.10 after you request. What's wrong with my logs?

dmitryfleytman, 2015-12-28:
1.0.8 in file name

nhancom3d, 2015-12-28:
@dmitryfleytman: sorry, this is my fault. It actually are logs of UsbDk 1.0.10.

dmitryfleytman, 2015-12-29:
Hello @nhancom3d,

Could you also do the same experiment with selective suspend disabled (see my comment above: https://github.com/daynix/UsbDk/issues/14#issuecomment-167518310).

Thanks,
Dmitry

nhancom3d, 2015-12-29:
Hi @dmitryfleytman,
I've done all your guides, include experiment with selective suspend disabled. Do you want me do more anything else? I am always ready to perform for you.
Thanks.

dmitryfleytman, 2015-12-29:
Thanks @nhancom3d, I need some time to analyse your logs.
Could you specify which logs were collected with enabled selective suspend and which with disabled?

nhancom3d, 2015-12-30:
Hi @dmitryfleytman,
All my log after your comment (https://github.com/daynix/UsbDk/issues/14#issuecomment-167518310) were collected with disabled selective suspend.
Previously, I set enabled selective suspend. After your guide I have set disabled selective suspend already. Is this OK? Do you want me collect log with "enabled selective suspend" case.
Thanks.

dmitryfleytman, 2015-12-30:
Hi @nhancom3d,

Yes, please collect the same log with selective suspend enabled. My intention is to see if there is any difference.

nhancom3d, 2015-12-30:
Hi @dmitryfleytman 
Sorry about the delay. I have experiment all scenarios (include dump Usb tree on on both the host + VM and run Gemalto SmartDiag Diagnosis Tool on VM) with 2 case selective suspend enabled and disable (on both the host and the VM). Here is my log: [token_Usbdk1.0.10_log_20151230.zip](https://github.com/daynix/UsbDk/files/74798/token_Usbdk1.0.10_log_20151230.zip). Hope it helpful with you.
Thanks.

nhancom3d, 2016-01-04:
Happy new year 2016, @dmitryfleytman  ! Best wishes for you !
Today I come back to work after the holiday. Have you come back to work yet?

dmitryfleytman, 2016-01-06:
Thanks, @nhancom3d . Happy new year to you as well! Yes, we're working already.
I'm looking into this issue. I believe I have some progress but it's too early to say something specific. I'll get back too you as soon as possible.

dmitryfleytman, 2016-01-14:
Hi, @nhancom3d,

I'm looking at your logs, there is something I'm missing in UsbTree dump from PC. Could it be that you created it when the usb device was redirected? Could you please uninstall UsbDk and dump UsbTree output again?

Thanks, Dmitry.

nhancom3d, 2016-01-14:
Hi, @dmitryfleytman,
I'm done before I create dump UsbTrace USB redirect. I tried to dump UsbTree after redirect USB, log obtained contain some information - I think it's useful for you. Therefore, I will reinstall UsbDk 1.0.10, repeat all the steps above test, adding the dump UsbTree after redirect USB. Please wait for me a few minutes (~15m) to collect log.

dmitryfleytman, 2016-01-14:
Thanks, @nhancom3d, please also do not forget to collect logs with UsbDk uninstalled.

nhancom3d, 2016-01-14:
Hi, @dmitryfleytman,
I send you my blog had collected. Including 2 cases:
- With_UsbDk: Repeat the above steps, adding the dump UsbTree after redirect USB.
- Without_UsbDk: Only include information dump UsbTree after uninstalling UsbDk.
[token_Usbdk1.0.10_log_20160114.zip](https://github.com/daynix/UsbDk/files/90298/token_Usbdk1.0.10_log_20160114.zip)

dmitryfleytman, 2016-01-14:
Hi @nhancom3d,

Are you running UsbTree on physical machine as administrator? For some reason its cannot fetch full device info for your device. If not, please run it as administrator and collect the log (no need to do play with selective suspend, it doesn't matter at this stage).

Namely, I miss string descriptors:
>       ---===>Device Information<===---
> *!*ERROR:  No String Descriptor for index 2!

> ...

> *!*ERROR:  No String Descriptor for index 1!
iProduct:                          0x02
> *!*ERROR:  No String Descriptor for index 2!
iSerialNumber:                     0x03
> *!*ERROR:  No String Descriptor for index 3!
bNumConfigurations:                0x01

And full configuration descriptor:

>        ---===>Full Configuration Descriptor<===---

Thanks,
Dmitry

nhancom3d, 2016-01-15:
Hi @dmitryfleytman 
I always run the application with administrator. I discovered this problem:
- When I plug the device for the first time after rebooting, performing dump UsbTree will obtain information Token device successfully.
- When I unplug the device and plug it in again, it does not obtain device information.

I tried with one other PC, the same thing happens.
Here is my log in 2 cases: 
[UsbTree_selective_suspend_enabled_Without_UsbDk_20150115.zip](https://github.com/daynix/UsbDk/files/91317/UsbTree_selective_suspend_enabled_Without_UsbDk_20150115.zip)

Thanks for support !

nhancom3d, 2016-01-15:
Hi, @dmitryfleytman 
I have made some tests. I discovered that after when unplug and plug Token device to another USB port,  performing dump UsbTree will obtain information Token device successfully, not need reboot PC. Errors occur when plug Token device again into old USB port. I have borrowed orther USB Token (same product line with my UsbToken), it also meets the same error. Perhaps this is the common error of this token usb product line.

dmitryfleytman, 2016-01-19:
Hi, @nhancom3d, does the same issue happen on VM, i.e. what if you redirect/disconnect/redirect again the device, is UsbTree able to read device information properly after that?

And on the host, after device was replugged into the same Usb port, does Gemalto tool recognise it?

nhancom3d, 2016-01-20:
Hi, @dmitryfleytman, on VM when redirect/disconnect/redirect again the device, UsbTree always can able to read device infomation. Even in case PC can't get device infomation. I did not detect any difference in the dump tree file of UsbTree tool in all cases.
On the host, after device was replugged into the same Usb port, Gemalto tool also recognise (pass) it, same other case.

dmitryfleytman, 2016-01-20:
Hi, @nhancom3d 

I sent you a private message to nhancom3d@gmail.com
Did you see it?

~Dmitry

nhancom3d, 2016-01-21:
Hi @dmitryfleytman,
Sorry about my delay. I has reply your mail, please check your inbox.
Thanks.

closed 2016-06-27 by nhancom3d