daynix/UsbDk #15: Redirect of USB device failed

terry666, 2015-12-09:
I have installed USBDK, but  I excute the command“UsbDkController.exe -r ID SN”in cmd on windows7(32 bit), the error "Driver file operation error. DeviceIoControl failed (The system cannot find thefile specified.  Error code = 2)"  will appear. How to solve this problem?
thank you！

        The cmd screenshot is following：

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -i
Installing UsbDk driver
UsbDk driver installation succeeded

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -n
Enumerate USB devices
Found 5 USB devices:
0. FilterID: 1, Port: 1, ID: 8087:0024, Configs: 1, Speed: 3
        USB\VID_8087&PID_0024 1
Descriptor for configuration #0: size 25
1. FilterID: 3, Port: 1, ID: 8087:0024, Configs: 1, Speed: 3
        USB\VID_8087&PID_0024 1
Descriptor for configuration #0: size 25
2. FilterID: 4, Port: 2, ID: 413c:2107, Configs: 1, Speed: 2
        USB\VID_413C&PID_2107 2
Descriptor for configuration #0: size 34
3. FilterID: 4, Port: 4, ID: 17ef:6019, Configs: 1, Speed: 2
        USB\VID_17EF&PID_6019 4
Descriptor for configuration #0: size 34
4. FilterID: 5, Port: 1, ID: 0951:1665, Configs: 1, Speed: 3
        USB\VID_0951&PID_1665 60A44C413C4EFEC1798423E8
Descriptor for configuration #0: size 32

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r 0951:1665 60A44C413C4EFEC1798423E8
Redirect USB device 0951:1665, 60A44C413C4EFEC1798423E8
Driver file operation error. DeviceIoControl failed (The system cannot find thefile specified.  Error code = 2)
Redirect of USB device failed

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r 0951:1665 60A44C413C4EFEC1798423E8
Redirect USB device 0951:1665, 60A44C413C4EFEC1798423E8
Driver file operation error. DeviceIoControl failed (The system cannot find thefile specified.  Error code = 2)
Redirect of USB device failed

dmitryfleytman, 2015-12-09:
Hello and thank you for your report.

The correct syntax for what you’re doing is 

UsbDkController.exe -r "USB\VID_0951&PID_1665" 60A44C413C4EFEC1798423E8

Hellp screen of UsbDkController.exe is indeed misleading, I will fix it in the next UsbDk version.

Best Regards,
Dmitry

closed 2015-12-10 by terry666

dmitryfleytman, 2015-12-28:
Help screen clarifications committed at 51ca12740b03237715bffb576c1d78635590b72e