daynix/UsbDk #16: How can i integrate backend UsbDk (replace WinUSB) on Usb-redirection features of spice-gtk

syshack, 2015-12-09:
My steps is follow,but doesn‘t work.
1、Download and install USBDK from http://www.spice-space.org/download/windows/usbdk/UsbDk_1.0.8_x86.msi on my laptop  and test with USBDKcontroller.exe -r，it works well
2、Clone https://github.com/dmitryfleytman/libusb/tree/usbdk-backend-v5 and compile
     mingw32-configure –enable-usbdk –enable-debug-log
     make 
     make install
3、compile spice-gtk-0.29 
     mingw32-configure –with-gtk=2.0 –enable-usbredir=yes –with-sasl=no –enable-vala 
     make 
     make install 
     cd spice-common/spice-protocol 
     make install 
 4、 compile  virt-viewer-1.0 
     mingw32-configure –with-gtk=2.0 –with-spice-gtk 
     make 
     make install 
 5、Package and install remote-viewer on my laptop 
 6、Copy UsbDkHelper.dll to the same dir of remoteviewer.exe
 7、open remote-viewer and connect to my vm，click the redirect checkbox, raise the error "-12" .

Is there any step by step document?

nhancom3d, 2015-12-09:
Hi syshack,
If you redirect usb with spice-gtk 0.29, it fail if you don't install usbclerk. If you using spice-gtk 0.28, it not check usbclerck is running or not.

dmitryfleytman, 2015-12-09:
This is a little bit complicated right now because a lot of things are in the air pending acceptance by libusb and spice-gtk upstream.

Strictly speaking, in addition to libusb patches you also need a patch set for spice-gtk:
http://lists.freedesktop.org/archives/spice-devel/2015-October/022851.html

I hope it will become easier soon when everything accepted upstream :)

syshack, 2015-12-10:
@nhancom3d @dmitryfleytman  Thank you all , for nhancom3d's help ,it works now,i'll do further test .

closed 2015-12-10 by syshack