daynix/UsbDk #17: Can't redirect JetFlash USB TS1GJFV30

nhancom3d, 2015-12-16:
Hi @dmitryfleytman 
I have a USB storage device named JetFlash USB TS1GJFV30 Device (USB \ VID_058F & PID_6387, 4SESX7O4). I run the command: `UsbDkController.exe -r  “USB \ VID_058F & PID_6387" 4SESX7O4`, it successfully. But when I try to redirect to the virtual machine, the VM does not find the device. I checked Device Manager and Events Log of VM, but no information. My environment end my step is similar to with issue #14. Please do check and respond to me soon on this issue.
Here is my UsbDk log files: 
[Usb_transcend_binaryLog_usbdk1.0.8.zip.txt](https://github.com/daynix/UsbDk/files/63644/Usb_transcend_binaryLog_usbdk1.0.8.zip.txt)

Additional Information: Previously, I also meet similar problems with this Usb device and UsbDk v1.0.7. UsbDk log notice "GetOverlappedResult failed" when plug this device.I fixed it by comment line 102 of file UsbDkHelper\RedirectorAccess.cpp for skip expection, then rebuild again UsbDK it has redirect OK.
```
throw UsbDkDriverFileException(TEXT("GetOverlappedResult failed"));
```
After upgrading to UsbDk 1.0.8, I do not see this log on again. Therefore I do not try solutions above.

dmitryfleytman, 2015-12-20:
Adding text log extracted from binary trace files.
(https://github.com/daynix/UsbDk/files/67736/UsbDkTrace_GitHub_Issue16.txt)

dmitryfleytman, 2015-12-20:
Adding text lof extrcted from binary trace files.
(https://github.com/daynix/UsbDk/files/67738/UsbDkTrace_GitHub_Issue17.txt)

dmitryfleytman, 2015-12-20:
Hi @nhancom3d,

I analysed your log and the root cause of the problem you observe is an attempt to reset a non-existing pipe of usb device.

> [1]0CB0.15C8::12/16/2015-06:52:25.972 [UsbDk]CWdfUsbTarget::ResetPipe Failed: Pipe not found

I've found a pretty similar device here, from the same family but 16Gb, and tried to reproduce the problem. I don't this problem and device is working as expected. Currently it is not clear who is responsible for this invalid pipe reset attempt, it may be a bug in UsbDk/libusb or a problem of the guest driver. Therefore, I need more information from your side:

1. What guest operation system are you running?
2. I prepared an internal build of UsbDk with extended trace logs to collect more information about the problem [1]. Could you please install it and collect the new log? Please pay attention that this build is not signed, so you will need to enable test signing mode as described in [2] in order to run it on 64 bit systems. 

[1] https://www.dropbox.com/s/dv0q9qkpcdxjvsx/Install_Debug.zip?dl=0
[2] https://github.com/daynix/UsbDk/blob/master/Documentation/DriverSigning.txt#L33

Thanks in advance,
Dmitry

nhancom3d, 2015-12-21:
Sorry @dmitryfleytman,
Today, I forgot that USB stick at home. I do not have same that device on my company; other devices work well with UsbDk. I will follow your instructions and send your detailed log after 2 days.
My guest operation system is Windows 7 32 bit, I don't need perform [2], right?

dmitryfleytman, 2015-12-21:
Yes, [2] is not needed for 32 bit hosts.

nhancom3d, 2015-12-24:
Hi @dmitryfleytman,
I'm so sorry for the delay. This is my log: 
[log_usbdk_1.0.8_transcend.zip](https://github.com/daynix/UsbDk/files/71367/log_usbdk_1.0.8_transcend.zip)
Thanks for support !

dmitryfleytman, 2015-12-28:
Hi @nhancom3d,

Would you please do a few more experiments with your device:

1. I’ve just released the new UsbDk version - 1.0.10 (https://github.com/daynix/UsbDk/releases/tag/v1.00-10). It contains extended tracing facilities so I’d like to ask you to collect new logs with this version, UsbDk will trace much more info about what is happening w.r.t. your device.

2. Please also do the same experiment (and collect logs) with USB selective suspend disabled on the host:

Go to control panel-->Power Options—>Edit plan settings->Change advanced power settings, under USB settings—>USB selective suspend setting change both "On battery”/“Plugged In” to disabled and reboot the system.

3. Besides that, please collect usb tree dump from your system using UsbView (on both the host and the VM):

* See https://github.com/daynix/UsbDk/blob/master/Documentation/UsbInfo.txt

Thanks,
Dmitry

nhancom3d, 2015-12-28:
Hi @dmitryfleytman 
I've followed your instructions and perform this scenario:
- Running UsbDkTraceToFile.bat file to start logging
- Plug the Transcend USB device, wait 30 seconds.
- Collect Usb tree dump on host
- Connect to a virtual machine using virt-viewer 2.0.
- Start redirect USB token device, wait 30 seconds
- Collect Usb tree dump on VM
- Stop redirect USB token and virt-viewer windows closed.
- Wait 30 seconds, ending logging and unplug usb device

This is my log I've collected ([transcend_log_20151228.zip](https://github.com/daynix/UsbDk/files/73186/transcend_log_20151228.zip)). I hope it contains enough information as you expect.

Thanks for support !

dmitryfleytman, 2016-01-14:
Parsed log for version 1.0.10:

[Issue17-1.0.10-log.txt](https://github.com/daynix/UsbDk/files/90178/Issue17-1.0.10-log.txt)

dmitryfleytman, 2016-01-14:
Hi @nhancom3d, thanks for the logs, they are helpful.
I succeeded to simulate and fix the issue you observe on my systems.

Please check if attached binaries work for you: 
[UsbDk_Debug_99.99.90_x64.msi.zip](https://github.com/daynix/UsbDk/files/90188/UsbDk_Debug_99.99.90_x64.msi.zip)
[UsbDk_Debug_99.99.90_x86.msi.zip](https://github.com/daynix/UsbDk/files/90189/UsbDk_Debug_99.99.90_x86.msi.zip)

nhancom3d, 2016-01-19:
Hi @dmitryfleytman 
It works with me. This bug has been fixed. Thank you very much !
Please commit code you added to fix this bug.
Thanks !

closed 2016-01-19 by nhancom3d

dmitryfleytman, 2016-01-19:
Very well! I'll push corresponding changes and release a new UsbDk version during this week.

dmitryfleytman, 2016-01-21:
Fixed at c573972ff10aa1da8f2a58ec220d22a2f3678712.

nhancom3d, 2016-01-22:
Great! Thanks @dmitryfleytman.