daynix/UsbDk #18: Wacom tablet cannot work, when redirected to a qemu-kvm vm using spice+usbredir+usbdk

xuyongjian, 2016-01-06:
I am using usbdk in spice-gtk to redirect a wacom table device.
I only replaced the libusb using libusb-usbdk-backend without modifying the spice-gtk.
It worked well with Mass Storage Device. However, some error occurred when inserted a tablet device. Some interrupt requests return error with Cancelled status.

xuyongjian, 2016-01-07:
[myusbtree-host.txt](https://github.com/daynix/UsbDk/files/80702/myusbtree-host.txt)
[myusbtree-host.xml.txt](https://github.com/daynix/UsbDk/files/80703/myusbtree-host.xml.txt)
[myusbtree-vm.txt](https://github.com/daynix/UsbDk/files/80704/myusbtree-vm.txt)
[myusbtree-vm.xm.txt](https://github.com/daynix/UsbDk/files/80705/myusbtree-vm.xm.txt)

Log attached~

dmitryfleytman, 2016-01-14:
UsbDk traces for the issue:
[Issue18-Wacom-etl.zip](https://github.com/daynix/UsbDk/files/90226/Issue18-Wacom-etl.zip)
[UsbDkTrace-suspend-disable.txt](https://github.com/daynix/UsbDk/files/90227/UsbDkTrace-suspend-disable.txt)
[UsbDkTrace-suspend-enable.txt](https://github.com/daynix/UsbDk/files/90228/UsbDkTrace-suspend-enable.txt)

dmitryfleytman, 2016-01-14:
Hi @xuyongjian , thanks for the logs, they are helpful.
After investigation, I came up with an idea of what might be the problem and did a fix that should help.

Please check if attached binaries work for you (and collect new UsbDk trace log if they don't): 
[UsbDk_Debug_99.99.90_x64.msi.zip](https://github.com/daynix/UsbDk/files/90230/UsbDk_Debug_99.99.90_x64.msi.zip)
[UsbDk_Debug_99.99.90_x86.msi.zip](https://github.com/daynix/UsbDk/files/90231/UsbDk_Debug_99.99.90_x86.msi.zip)

Thanks,
Dmitry

xuyongjian, 2016-01-17:
Hi Dmitry:
Unfortunately, the tablet did not work. 
It info me that "Unfortunately, we don’t support that file type", while I was uploading a zip file......
I have sent the log to your email. Please help me upload it :)

Thanks
Yongjian

dmitryfleytman, 2016-01-19:
The new log: 
[UsbDkTrace.zip](https://github.com/daynix/UsbDk/files/96018/UsbDkTrace.zip)

dmitryfleytman, 2017-07-25:
Should be fixed in v1.0.19.
Please retest with the latest build and re-open in case problem still exists.

closed 2017-07-25 by dmitryfleytman

xuyongjian, 2017-10-11:
Tested with v1.0.19 but problem still existed in XP.
Worked fine in Win7.