daynix/UsbDk #19: Kinect v2 not found

xlz, 2016-01-27:
I was trying libusb with the new usbdk backend on Windows 8.1. But usbdkcontroller.exe was unable to find the Kinect. `usbdkcontroller.exe -n` found two devices of the Kinect adapter (Hub devices), but the Kinect sensor was not found. I could see the Kinect sensor device in the Windows device manager, and a Microsoft driver is installed for it.

What can I do?

dmitryfleytman, 2016-01-28:
Hi,

Please collect binary UsbDk trace for kinect connection as described here: https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52

Scenario should be:

1. Unplug your kinect
2. Run tracing
3. Plug the kinect back

Also please provide your USB topology (with kinect connected) as described here: https://github.com/daynix/UsbDk/blob/master/Documentation/UsbInfo.txt

Thanks,
Dmitry

xlz, 2016-01-29:
[usbdkcontroller-n.txt](https://github.com/daynix/UsbDk/files/110179/usbdkcontroller-n.txt)
[MyUsbTree.txt](https://github.com/daynix/UsbDk/files/110180/MyUsbTree.txt)
[MyUsbTree.xml.txt](https://github.com/daynix/UsbDk/files/110182/MyUsbTree.xml.txt)
[UsbDkTrace.etl.txt](https://github.com/daynix/UsbDk/files/110183/UsbDkTrace.etl.txt)

(Remove .txt suffix from the .xml and .etl files)

dmitryfleytman, 2016-01-31:
Thanks for the logs. We will do the investigation.

dmitryfleytman, 2016-02-10:
Fixed in v1.00-12 by 6b21ddc78cb53e6885b67ca6ee68975a1db645aa

closed 2016-02-10 by dmitryfleytman

xlz, 2016-02-10:
I confirm the Kinect is now detected.