daynix/UsbDk #20: USB keyboard doesn't work after installing UsbDK

xlz, 2016-02-26:
I installed UsbDk on a native Windows 7 machine. After reboot, the USB keyboard is no longer functioning. As you can imagine, accessing the machine is quite difficult now without a keyboard.

I'm still trying to remove the UsbDk driver from recovery mode where there is no access to the Device Manager.

Is UsbDk somehow used as the exclusive driver for the keyboard?

dmitryfleytman, 2016-02-27:
Hi!

No, UsbDk is not used as exclusive driver for the keyboard, it is a filter driver.

The best option to re-gain control over the system is connect to it from another computer using remote desktop and uninstall UsbDk.

According to what you describe UsbDk driver was not loaded properly so USB stack was not initialised. Could you please open device manager and check is USB host controller driver Okay and what is the problem if it's not (inspect USB host controller properties from right-click context menu).

One known issue with UsbDk driver on Windows 7 is that there is a problem with SHA-2 code signing certificates. In some cases SHA-2 signed drivers fail to load on these systems while the signature is valid and may be verified successfully with signtool.exe. Microsoft fixed this issue by security update 3033929. This update needs to be downloaded and installed in order to load such drivers.
The update may be downloaded from https://technet.microsoft.com/en-us/library/security/3033929.

Please see if this is your case.

xlz, 2016-02-27:
Unfortunately remote desktop was not enabled on the machine.

I wiped and reinstalled Windows 7 on the machine, enabled remote desktop, and reproduced the error.
[image]

When I ran the UsbDK installer, it got stuck at 20%, then after a while it prompted for something like "restart computer to finish the installation".

Indeed, I used the remote desktop to install security update 3033929 and that fixed the issue.

The system is stock Windows 7 sp1. Hopefully the UsbDk installer comes up with a way to check this requirement when installing so all this can be avoided in the first place.

closed 2016-02-27 by xlz

flavioroth, 2016-08-23:
Same issue as @xlz on Windows 10 after compiling and install the driver. Any idea what may cause this issue ?

dmitryfleytman, 2016-08-23:
Did you enable test signing on your system?

flavioroth, 2016-08-23:
Prior to loading the driver, yes.

dmitryfleytman, 2016-08-23:
What is the error message in device manager's properties of USB host controller? Are you installing debug or release build? Please note that release builds are unsigned by default.

dmitryfleytman, 2016-08-23:
Could you also try to install from official MSI?

dawguy, 2017-02-12:
I ran into this issue yesterday when installing from the latest x64 MSI. A signature error popped up and after rebooting my keyboard and mouse would no longer work. I ended up reinstalling Windows 10 from a USB to get them to work again. Its very likely the same issue the other users had a year ago.

Is there any way you can check whether or not test signing is enabled before you install and prevent the installation if it isn't?

dmitryfleytman, 2017-02-13:
Hi @dawguy,

I'm sorry to hear that you had to reinstall Windows due to UsbDk load failure.
Did you try to boot in safe mode and uninstall UsbDk?

A few questions:

1. Did you install UsbDk from officially released MSI or built it yourself?
In case of officially released MSIs there is no need of testsigning mode because drivers there are signed by a production signature. The issue users had year ago was specific to Windows 7 and absence of a SHA2 signatures support fix. On Windows 10 there should be no such a problem.

2. Are you using, by any chance, Windows 10 RS1 with secure boot enabled?
Clean installation (not upgrade) of Windows 10 RedStone1 on machines with secure boot enabled requires drivers cross-signed by Microsoft, UsbDk drivers do not have this signature yet.

Thanks,
Dmitry