daynix/UsbDk #21: Kinect 2 not found on Windows 7

xlz, 2016-02-28:
Similar to #19.

Files:
[UsbDkController-n.txt](https://github.com/daynix/UsbDk/files/149662/UsbDkController-n.txt)
[MyUsbTree.txt](https://github.com/daynix/UsbDk/files/149663/MyUsbTree.txt)
[UsbDkTrace.etl.txt](https://github.com/daynix/UsbDk/files/149664/UsbDkTrace.etl.txt)
(usbview.exe crashes when saving to xml.)

dmitryfleytman, 2016-02-28:
What version of UsbDk are you using?

xlz, 2016-02-28:
UsbDk_1.0.12_x64

dmitryfleytman, 2016-02-28:
Text extracted from .etl file:

[UsbDk]<lambda_a257bb3effc18ee15b227ef0411f37ef>::operator() Starting relations array processing:
[UsbDk]CDeviceRelations::Dump Array size: 1 (ptr: FFFFFA8009B78B00)
[UsbDk]CDeviceRelations::ForEachIf<bool #0: FFFFFA80067B1570
[UsbDk]UsbDkSendUrbSynchronously Send URB IRP Error 0xc000000d(STATUS_INVALID_PARAMETER)
[UsbDk]CUsbDkHubFilterStrategy::RegisterNewChild Cannot query device descriptor
[UsbDk]<lambda_a257bb3effc18ee15b227ef0411f37ef>::operator() Finished relations array processing
[UsbDk]<lambda_a257bb3effc18ee15b227ef0411f37ef>::operator() Starting relations array processing:
[UsbDk]CDeviceRelations::Dump Array size: 2 (ptr: FFFFFA8007757B80)
[UsbDk]CDeviceRelations::ForEachIf<bool #0: FFFFFA800ACC1600
[UsbDk]CDeviceRelations::ForEachIf<bool #1: FFFFFA80067B1570
[UsbDk]UsbDkSendUrbSynchronously Send URB IRP Error 0xc000000d(STATUS_INVALID_PARAMETER)
[UsbDk]CUsbDkHubFilterStrategy::RegisterNewChild Cannot query device descriptor
[UsbDk]UsbDkSendUrbSynchronously Send URB IRP Error 0xc000000d(STATUS_INVALID_PARAMETER)
[UsbDk]CUsbDkHubFilterStrategy::RegisterNewChild Cannot query device descriptor
[UsbDk]<lambda_a257bb3effc18ee15b227ef0411f37ef>::operator() Finished relations array processing
[UsbDk]UsbDkEvtDeviceAdd Entry
[UsbDk]CUsbDkFilterDevice::Create Entry
[UsbDk]CWdfDevice::CacheDeviceName Newly created device name is \Device\UsbDkFilter18
[UsbDk]CRegText::Dump ID: USB\VID_045E&PID_02D9
[UsbDk]CUsbDkFilterDevice::CStrategist::SelectStrategy Cannot query cached device descriptor
[UsbDk]CUsbDkFilterDevice::DefineStrategy Not attached
[UsbDk]UsbDkEvtDeviceAdd Exit 0xc00000bb(STATUS_NOT_SUPPORTED)
[UsbDk]CUsbDkFilterDevice::ContextCleanup Entry

dmitryfleytman, 2016-02-28:
According to the trace file device does not provide device descriptors. What is the difference since the last time when it worked?

xlz, 2016-02-28:
Last time it was Windows 8.1 with intel xhci. This is NEC controller on Windows 7.

libusb with libusbK backend works though.

xlz, 2016-02-29:
Anything can I do or any more data can I provide?

dmitryfleytman, 2016-03-01:
Not for now, thanks. I will take a look at this as soon as possible.

dmitryfleytman, 2017-02-02:
Duplicate of #33
Fixed by db95b4651e079e59353d8bafa84cc5ad23c4f2b1

closed 2017-02-02 by dmitryfleytman