daynix/UsbDk #22: I can't install UsbDk_1.0.12_x64.msi again!

jiaojialulu, 2016-03-06:
Hi
At first ,I uninstall UsbDk_1.0.12_x64.msi.
And then I want to use it again, but there is a warning like this:
[image]

dmitryfleytman, 2016-03-06:
Hi,

Thanks for the report.

Please try to do the same manually without MSI to see what is the problem.
Having UsbDk installed, run from UsbDk installation folder (c:\program files\UsbDk runtime libraries):
1. "UsbDkController.exe -u"
2. "UsbDkController -i"

Thanks,
Dmitry

jiaojialulu, 2016-03-06:
I can't install UsbDk,so I don't have this folder.

dmitryfleytman, 2016-03-06:
Oh, I see.
Could you check if you have files UsbDk.sys or UsbDkHelper.dll somewhere on the disk?

jiaojialulu, 2016-03-06:
I restart my computer some times, now it's OK!!!

closed 2016-03-06 by dmitryfleytman

lz2004xxw, 2020-09-10:
> I restart my computer some times, now it's OK!!!
> thanks !!!

how did u fixed it? i met the same problem