daynix/UsbDk #23: redirect usb device timeout

jinyyu, 2016-05-25:
hello.
I redirect my usb device using spice-gtk+libusb+usbdk, but it timeout, How to solve this problem?
thank you！

[UsbDk.txt](https://github.com/daynix/UsbDk/files/281438/UsbDk.txt)

jinyyu, 2016-05-27:
I have solved this problem, thanks

closed 2016-05-27 by jinyyu

dmitryfleytman, 2016-05-29:
Great! Could you please post a few words regarding what was the problem and how it was fixed?

Thanks,
Dmitry

jinyyu, 2016-06-02:
hello
Mybe it is just incomfortable for win10-64bit.
I try to use usbdk in win10, some devies works well while some devies redirect timeout.
I find it works well in win7.
thanks

dmitryfleytman, 2016-06-07:
Hi,

We didn't port/test UsbDk on Windows 10 yet. Please let us know if the same issue happens on pre-Win10 OS versions.

~Dmitry