daynix/UsbDk #26: windows10 probrem

yoshiki-k, 2016-06-13:
In an environment of Windows10 64bit(version 1511), usbdkcontroller fails to driver update of specific scanner.
It was successful in an environment of Windows7 and 8.1.

To see the log, I installed to compile the source of the same version as the UsbDk_1.0.12_x64.msi.
I tried to redirect the scanner using the usbdkcontroller.
However, the following errors has failed is displayed after 2 minutes.

[usbdkcontroller]
Driver file operation error. DeviceIoControl failed (The device is not connected.  Error code = 1167) Redirect of USB device failed

Try the other device, there was no problem.
*other usb2.0 scanner
*usb3.0/2.0 usb memory
*usb1.1 headset

In the debug trace log, there was a following of log immediately after redirect the scanner.

[debug trace]
00000140	10.97756100	UsbSleepStudy_GetDeviceStringProperty: Failed to fetch DeviceProperty 0xc0000034 

windows has failed to detect the redirected device.

I made a modification to perform the IOCTL_INTERNAL_USB_CYCLE_PORT ioctl only, but it was successful.

[logfile]
[UsbDkController.txt](https://github.com/daynix/UsbDk/files/311440/UsbDkController.txt)
[MyUsbTree.txt](https://github.com/daynix/UsbDk/files/311441/MyUsbTree.txt)
[MyUsbTree.xml.txt](https://github.com/daynix/UsbDk/files/311315/MyUsbTree.xml.txt)
[UsbDkTrace.txt](https://github.com/daynix/UsbDk/files/311442/UsbDkTrace.txt)

you need me to provide further information to diagnose this problem?

Regards,
 yoshiki-k

dmitryfleytman, 2016-06-14:
Hi, Thanks for the report.

Could you please post the full model name of the device you're working with (your scanner).

Thanks,
Dmitry

yoshiki-k, 2016-06-14:
The name of the scanner I'm working is 
"FUJITSU Image Scanner fi-7160".

Regards,
yoshiki-k

yoshiki-k, 2016-06-15:
I'm sorry, the scanner was not on the list to MyUsbTree file.
Upload again.

[MyUsbTree.txt](https://github.com/daynix/UsbDk/files/315687/MyUsbTree.txt)
[MyUsbTree.xml.txt](https://github.com/daynix/UsbDk/files/315688/MyUsbTree.xml.txt)

Regards,
 yoshiki-k

yoshiki-k, 2016-06-24:
This scanner does not include the device name in the USB descriptor.
A problem that the OS fails to detect the device, might be due to the fact that does not contain a device name to the USB descriptor.

In kernel debugging,it had to return the following errors in following the response of the IRP.
IRP_MN_QUERY_DEVICE_TEXT (DeviceTextDescription) 0xc00000bb (STATUS_NOT_SUPPORTED)
IRP_MN_QUERY_DEVICE_TEXT (DeviceTextLocationInformation) STATUS_SUCCESS

For Windows10 host controller on the environment of, EHCI will succeed, XHCI will fail.
win10 XHCI fail
win10 EHCI success
win8.1 XHCI success
win8.1 EHCI success

PDO driver is wrong?

yoshiki-k

dmitryfleytman, 2016-06-27:
Or UsbDk uses it in a wrong way...

yoshiki-k, 2016-06-29:
The audio device of a cypress was also failed in windows10.
device descriptor of the device is as follows.

--- Device Descriptor ---
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0110
bDeviceClass:                      0x00  -> This is an Interface Class Defined Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x08 = (8) Bytes
idVendor:                        0x04B4 = Cypress Semiconductor
idProduct:                       0x0306
bcdDevice:                       0x0110
iManufacturer:                     0x00
iProduct:                          0x00
iSerialNumber:                     0x00
bNumConfigurations:                0x01

I think, the problem is that iProduct is not defined (which is common with the scanner ).

Fix the source code of usbdk.
 - After usbdk has transferred IRP_MN_QUERY_DEVICE_TEXT (DeviceTextDescription) to PDO driver,
  If an error is returned, it is set to "usbdk device" name.

As a result, Windows has found a device (VID_2B23 & PID_CAFE) immediately after the redirect.

Attach the patch.
[diff.txt](https://github.com/daynix/UsbDk/files/338598/diff.txt)

Regards,
 yoshiki-k

dmitryfleytman, 2016-06-29:
Hi, the patch looks good, however I cannot understand the root cause of the problem.

UsbDk does not touch device text descriptors while redirecting. It changes VID/PID only.
This means that in case device is recognised by Windows without UsbDk, it should be recognised when UsbDk redirects it as well.
In other words - if someone fails to query text device descriptor with UsbDk - he will fail to do it without UsbDk as well.

Is your device working ok without UsbDk?

Do you know where

"00000140 10.97756100 UsbSleepStudy_GetDeviceStringProperty: Failed to fetch DeviceProperty"

printout comes from?

yoshiki-k, 2016-06-30:
>Is your device working ok without UsbDk?
If you do not use the usbdk, it works fine.

>Do you know where
>"00000140 10.97756100 UsbSleepStudy_GetDeviceStringProperty: Failed to fetch DeviceProperty"
>printout comes from?
It has been printed in UsbDkTrace, but this is probably the output of USBHUB3.sys.

When I insert the device, OS will correctly recognize the device.

--- device insert ---
00000009	0.00622089	UsbSleepStudy_GetDeviceStringProperty: Enter
00000010	0.00622738	UsbSleepStudy_GetDeviceStringProperty: Device 0XFFFFE00092A586C0
00000011	0.00623278	UsbSleepStudy_GetDeviceStringProperty: DeviceProperty 0000000000
00000012	0.00633433	UsbSleepStudy_GetDeviceStringProperty: PropertyString fi-7160
00000013	0.00633866	UsbSleepStudy_GetDeviceStringProperty: Exit
00000014	0.00634244	UsbSleepStudy_GetDeviceStringProperty: Enter
00000015	0.00634838	UsbSleepStudy_GetDeviceStringProperty: Device 0XFFFFE00092A586C0
00000016	0.00635324	UsbSleepStudy_GetDeviceStringProperty: DeviceProperty 0x00000001
00000017	0.00640348	UsbSleepStudy_GetDeviceStringProperty: PropertyString USB\VID_04C5&PID_132E&REV_0100
00000018	0.00640780	UsbSleepStudy_GetDeviceStringProperty: Exit
00000019	0.00641320	UsbSleepStudy_RegisterPdo: New pCsContext DeviceDescription fi-7160
00000020	0.00641969	UsbSleepStudy_RegisterPdo: New pCsContext HardwareId USB\VID_04C5&PID_132E&REV_0100
00000021	0.00642455	UsbSleepStudy_RegisterPdo: Exit 0000000000
00000022	0.00644778	UsbSleepStudy_DeviceD0Entry: Enter
00000023	0.00645372	UsbSleepStudy_DeviceD0Entry: Handle 0XFFFFE00092DF3A20
00000024	0.00645966	UsbSleepStudy_DeviceD0Entry: Exit STATUS_SUCCESS Context 0XFFFFE00092DF3A20
00000025	0.00646560	UsbSleepStudy_DeviceD0Entry: Exit Handle 0XFFFFE00092DF3A20

When redirect the device, Will try to re-recognition to reset the device,
In via usbdk filter driver, in effect changing the VID / PID, OS will fail to get the device name.

--- usbdk redirection start ---
00000140	10.97756100	UsbSleepStudy_GetDeviceStringProperty: Failed to fetch DeviceProperty 0xc0000034

The registry, the following information has been recorded.

[Original Device Information]
HKLM\SYSTEM\CurrentControlSet\Enum\USB\VID_04C5&PID_132E\5&28677613&0&12
  DeviceDesc = @oem55.inf,%usb\fujitsu_fi-7160u______0.devicedesc%;fi-7160
  HardwareID = USB\VID_04C5&PID_132E&REV_0100 USB\VID_04C5&PID_132E
  ...

[Case the OS can not be recognized by the redirect]
HKLM\SYSTEM\CurrentControlSet\Enum\USB\Vid_2B23&Pid_CAFE&Rev_0001\5&28677613&0&UsbDk10
  (DeviceDesc is not present.)
  HardwareID = USB\Vid_2B23&Pid_CAFE&REV_0001 USB\Vid_2B23&Pid_CAFE
  ...

[Case of applying the patch]
HKLM\SYSTEM\CurrentControlSet\Enum\USB\Vid_2B23&Pid_CAFE&Rev_0001\5&28677613&0&UsbDk11
  DeviceDesc = usbdk device
  HardwareID = USB\Vid_2B23&Pid_CAFE&REV_0001 USB\Vid_2B23&Pid_CAFE
  ...

[Case of devices that have the iProduct to descriptor]
HKLM\SYSTEM\CurrentControlSet\Enum\USB\VID_04C5&PID_1200\5&28677613&0&5
  DeviceDesc = @oem33.inf,%usb\fujitsu_scansnaps1100____0.devicedesc%;ScanSnap S1100
  HardwareID = USB\VID_04C5&PID_1200&REV_0100 USB\VID_04C5&PID_1200
  ...

HKLM\SYSTEM\CurrentControlSet\Enum\USB\Vid_2B23&Pid_CAFE&Rev_0001\5&28677613&0&UsbDk11
  DeviceDesc = ScanSnap S1100
  HardwareID = USB\Vid_2B23&Pid_CAFE&REV_0001 USB\Vid_2B23&Pid_CAFE
  ...

When going through the filter driver, just unlike when inserted, OS does not seem to get the device name from inf file.
Therefore, devices that do not have the iProduct to descriptor seems DeviceDesc the registry becomes no longer be set.

When usbdk redirects in Win8.1, or in the EHCI host controller environment of Win10, OS will recognize as an "Unknown device".
 (Even if the registry does not have a DeviceDesc)

Windows10 and OS at the time of the combination of XHCI host controller does not recognize the device.
Behavior in the USBHUB3.sys and WDF1000.sys driver might have changed.

Regards,
 yoshiki-k

dmitryfleytman, 2016-06-30:
Ok, now I see. So the root cause is that UsbDk device does not have INF and some bus drivers do not like this.

I'd like to make a few amendments for your patch and then I'll post it here for you to test.

yoshiki-k, 2016-06-30:
Thank you, I will wait for it.

Thanks,
 yoshiki-k

dmitryfleytman, 2016-07-03:
Hi, please give try to attached patches.

I thought it will be better to patch device description text always, even if it was retrieved successfully.
In this case device will always look like "UsbDk device" in device manager.

BTW, do these patches fix #27 as well?

[UsbDk-Issue26-Patches.zip](https://github.com/daynix/UsbDk/files/345019/UsbDk-Issue26-Patches.zip)

yoshiki-k, 2016-07-04:
Thank you.
This patch worked correctly.

>BTW, do these patches fix #27 as well?

Because the device name has been fixed, there is no more need for corresponding for  #27 issues.

When is being officially released?

Regards,
 yoshiki-k

dmitryfleytman, 2016-07-04:
Great!
I'm adding patches into v1.0.14 that will be released in a next few days.

dmitryfleytman, 2016-07-04:
BTW, do you want to provide your "signed-off-by:" line for these patches?

yoshiki-k, 2016-07-05:
Thanks for the offer, but no thanks.

 yoshiki-k

dmitryfleytman, 2016-07-06:
Fixed in v1.00-14 by commit e68f982ff7370f9c75bcabce1b76f3df6d9299be.

closed 2016-07-06 by dmitryfleytman