daynix/UsbDk #27: Different device name is displayed in the device manager

yoshiki-k, 2016-06-29:
usbdk is using a sequence number which is counted from the time of OS start-up in the device information.
When I redirect the device in UsbDkController.exe, connection information of the device to the local PC registry will be logged.

Connect the device A to the root hub, after the redirect 10 times, and then restart the OS.
When I was redirected to connect the device B to the same port, the name of the device A has been displayed in the tree of other devices.
After it redirects 10 times, 11 times and later will show the correct name of the device B.

For the device that defines the iSerial in the device descriptor, it will be displayed in the previous device information regardless of the hub to connect.

Regards,
 yoshiki-k

dmitryfleytman, 2016-06-29:
Hi,

Not sure I understand what is the issue.

What do you mean by sequence number? UsbDk patches VID/PID only, device serial number left as is. Does this introduce any problem?

What application are you using to view the device tree. Could you please provide some screenshots?

Thanks,
Dmitry

yoshiki-k, 2016-06-30:
Thank you for looking at the post.
Is a screen shot.
[screenshot.zip](https://github.com/daynix/UsbDk/files/341264/screenshot.zip)

- Image of the first time usbdk redirects the "USB 2.0 Camera"
usbcamera1.png
registory1.png

- Image at the time was to redirect the "CMedia" but that has been displayed "USB 2.0 Camera"
cmedia1.png

- When the device instance path has become "USBDK12", "CMedia" is displayed correctly
cmedia2.png

Current registry is as follows:
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\Vid_2B23&Pid_CAFE&Rev_0001
  \5&28677613&0&UsbDk6  -> DeviceDesc = C-Media USB Headphone Set
  \5&28677613&0&UsbDk8  -> DeviceDesc = USB 2.0 Camera
  \5&28677613&0&UsbDk10 -> DeviceDesc = USB 2.0 Camera
  \5&28677613&0&UsbDk12 -> DeviceDesc = C-Media USB Headphone Set

[Operating procedure]

1. redirect "USB 2.0 Camera"
 "UsbDk8" key has been created.
 "USB 2.0 Camera" is displayed in the Device Manager.

2. redirect "USB 2.0 Camera"
 "UsbDk10" key has been created.
 "USB 2.0 Camera" is displayed in the Device Manager.

3. reboot windows

4. redirect "C-Media USB Headphone Set"
 "UsbDk6" key has been created.
 "C-Media USB Headphone Set" is displayed in the Device Manager.

5. redirect "C-Media USB Headphone Set"  (Wrong case)
 "UsbDk8" key was reused.
 "USB 2.0 Camera" is displayed in the Device Manager.

6. redirect "C-Media USB Headphone Set"  (Wrong case)
 "UsbDk10" key was reused.
 "USB 2.0 Camera" is displayed in the Device Manager.

7. redirect "C-Media USB Headphone Set"
 "UsbDk12" key has been created.
 "C-Media USB Headphone Set" is displayed in the Device Manager.

Regards,
 yoshiki-k

dmitryfleytman, 2016-07-06:
Fixed in v1.00-14 by commit e68f982ff7370f9c75bcabce1b76f3df6d9299be.

closed 2016-07-06 by dmitryfleytman