daynix/UsbDk #28: CWdmUsbDeviceAccess::GetDeviceDescriptor may block the system due to no return of the GET_DESCREPTOR request.

xuyongjian, 2016-08-23:
Should we set a timeout in CIoControlIrp::SendSynchronously() function?

dmitryfleytman, 2016-08-23:
@xuyongjian, could you please deceive scenario when this happens? What is the problematic device? What is the descriptor type? What is device state when this happens?

xuyongjian, 2016-08-23:
I install usbdk in a virtual machine powered by qemu & kvm.
The redirected device is "HP LaserJet M1319f MFP" printer.
When happened, the code was running into "CUsbDkFilterDevice::CStrategist::SelectStrategy". And blocked when calling UsbDkWdmUsbDeviceIsHub, which will send a request as I mentioned before.

dmitryfleytman, 2016-08-23:
Do you have any idea why this descriptor request did not return at the first place?

xuyongjian, 2016-08-23:
Not yet~
I catched usb log  on the host PC. Every usb packet seemed to be well.
Maybe it's a problem of the vm?

xuyongjian, 2016-08-23:
Maybe the device does not support to get descriptor many times?

dmitryfleytman, 2016-08-23:
According to USB standards this should not happen.

xuyongjian, 2016-08-25:
I removed the hub check, since I would not redirect a hub into the vm :)

dmitryfleytman, 2017-02-02:
Corresponding UsbDk logic was redesigned and now it requests device descriptor only once. This should solve the problem.

closed 2017-02-02 by dmitryfleytman