daynix/UsbDk #29: When 2 USB devices remove each USB hub, it can no longer be redirected.

yoshiki-k, 2016-08-24:
usbdk was successfully deleted PDO from list when only one of USB Device removed,
but case of two devices, it was not deleted.

I also attempt to once again redirect the USB devices, it will not be able to redirect because the PDO remains in the list.

Is it possible to fix this issue?

Regards,
 yoshiki-k

yoshiki-k, 2016-08-25:
I unplug the redirected device with USB hub.
I tried to once again redirect the device, but it was an error.

[Operating procedure]
1. connect USBHUB(VID_05E3&PID_0608) & USBMemory(VID_04BB&PID_0CCB)
2. redirect USBMemory
3. disconnect USBHUB(with USBMemory)
4. press return key (exit UsbDkController.exe)
5. connect USBHUB(VID_05E3&PID_0608) & USBMemory(VID_04BB&PID_0CCB)
6. redirect USBMemory
   Redirect of USB device failed

When I unplugged only device from the USB hub, the following log was written.
[UsbDk]CUsbDkRedirection::NotifyRedirectionRemovalStarted Redirector removal started for USB\VID_04BB&PID_0CCB
[UsbDk]CUsbDkControlDevice::GetPDOByDeviceID PDO was not found	

When I unplug the device in conjunction with a USB hub, the following log was written.
[UsbDk]CUsbDkRedirection::NotifyRedirectionRemovalStarted Redirector removal started for USB\VID_04BB&PID_0CCB
[UsbDk]CWdmUsbDeviceAccess::Reset Send IOCTL IRP Error 0xc000000e(STATUS_NO_SUCH_DEVICE)

Removed the device seem to remain in the list.

I've modified the source in order to delete the list of the redirected device.
Although the next redirect worked fine, it seems like this modification is not appropriate.

[ControlDevice.cpp]
NTSTATUS CUsbDkControlDevice::RemoveRedirect(const USB_DK_DEVICE_ID &DeviceId)
{
...
        //    return res;	// line 947 commented
...
}

Thank you for your advice.

[usbdkcontroller.txt](https://github.com/daynix/UsbDk/files/436833/usbdkcontroller.txt)
[traceview.txt](https://github.com/daynix/UsbDk/files/436836/traceview.txt)

Regards,
yoshiki-k

dmitryfleytman, 2016-08-26:
Hello and thank you for this report. I will investigate this.

dmitryfleytman, 2016-09-01:
Hi @yoshiki-k,

I made a fix and pushed it to branch d-fix-hub-removal (https://github.com/daynix/UsbDk/tree/d-fix-hub-removal). Could you please verify it on your system.

Thanks,
Dmitry

yoshiki-k, 2016-09-02:
UsbDk has been improved by this version!
I think that need to be fixed for spicy.

I expect the official release!

Thanks,
yoshiki-k

dmitryfleytman, 2016-09-04:
Fixed by bf79d6e9425b1587aaa542c31912f5f6a574cd64.

dmitryfleytman, 2016-09-04:
Fix will be included into v v1.00-15

closed 2016-09-04 by dmitryfleytman