daynix/UsbDk #31: Does not contain the catalog file

yoshiki-k, 2016-09-13:
Does not contain the catalog file in the release version of the driver msi, Can it be included?
Looking at the installed driver in Device Manager, Driver is not signed.

Regards,
yoshiki-k

dmitryfleytman, 2016-09-13:
Do you mean officially released MSI or custom built MSI?

dmitryfleytman, 2016-09-13:
In general there is no .cat file for UsbDk driver.

UsbDk.sys contains embedded RedHat signature for official builds, test signature for unofficial/custom Debug builds. Unofficial/custom Release builds are not signed.

Custom Release builds may be signed by test/real builder certificate by changing corresponding settings in Visual Studio project configuration.

closed 2016-09-13 by dmitryfleytman