daynix/UsbDk #33: FTDI 2232H based devices don't show up

tsailer, 2017-01-31:
FTDI 2232H based devices (VID 0403 PID 6010) don't show up in UsbDkController.exe -n

The following is a trace from when attaching the FTDI device. I fear it's quite empty...
https://sailer.fedorapeople.org/trace.zip

This is with UsbDk_1.0.15_x64

dmitryfleytman, 2017-01-31:
The log is not empty:

> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]UsbDkEvtDeviceAdd Entry
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]CUsbDkFilterDevice::Create Entry
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]CWdfDevice::CacheDeviceName Newly created device name is \Device\UsbDkFilter36
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]CRegText::Dump ID: USB\VID_0403&PID_6010
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]CUsbDkFilterDevice::CStrategist::SelectStrategy Cannot query cached device descriptor
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]CUsbDkFilterDevice::DefineStrategy Not attached
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]UsbDkEvtDeviceAdd Exit 0xc00000bb(STATUS_NOT_SUPPORTED)
> [0]0004.0058::01/31/2017-14:18:04.740 [UsbDk]CUsbDkFilterDevice::ContextCleanup Entry
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]UsbDkEvtDeviceAdd Entry
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]CUsbDkFilterDevice::Create Entry
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]CWdfDevice::CacheDeviceName Newly created device name is \Device\UsbDkFilter37
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]CRegText::Dump ID: USB\VID_0403&PID_6010&MI_00
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]CUsbDkFilterDevice::CStrategist::SelectStrategy Cannot query cached device descriptor
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]CUsbDkFilterDevice::DefineStrategy Not attached
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]UsbDkEvtDeviceAdd Exit 0xc00000bb(STATUS_NOT_SUPPORTED)
> [2]0004.0058::01/31/2017-14:18:04.777 [UsbDk]CUsbDkFilterDevice::ContextCleanup Entry
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]UsbDkEvtDeviceAdd Entry
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]CUsbDkFilterDevice::Create Entry
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]CWdfDevice::CacheDeviceName Newly created device name is \Device\UsbDkFilter38
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]CRegText::Dump ID: USB\VID_0403&PID_6010&MI_01
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]CUsbDkFilterDevice::CStrategist::SelectStrategy Cannot query cached device descriptor
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]CUsbDkFilterDevice::DefineStrategy Not attached
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]UsbDkEvtDeviceAdd Exit 0xc00000bb(STATUS_NOT_SUPPORTED)
> [2]0004.0058::01/31/2017-14:18:04.781 [UsbDk]CUsbDkFilterDevice::ContextCleanup Entry

dmitryfleytman, 2017-01-31:
@tsailer, the log is short indeed. Is it a log of device attach (physical plug-in) or enable via device manager?

tsailer, 2017-01-31:
This is a physical plug in. The device is already configured and Windows just loads the FTDI driver

dmitryfleytman, 2017-01-31:
I don't see bus re-enumeration events. It looks like UsbDk was not properly installed. Did you use MSI package to install it? Did installer ask to reboot the system after installation?

tsailer, 2017-01-31:
I did use MSI. And I rebooted (I think it asked, but I did anyway). Then when it didn't work I did subdkcontroller -u and then usbdkcontroller -i

dmitryfleytman, 2017-01-31:
Could you please check in regedit, that
HKLM\\System\\CurrentControlSet\\Control\\Class\\{36FC9E60-C465-11CF-8056-444553540000}\\ UpperFilters contains string "UsbDk"? Is there anything else except UsbDk?

Also, what is your USB controller? Could you post dump of UsbView output (https://github.com/daynix/UsbDk/blob/master/Documentation/UsbInfo.txt)

tsailer, 2017-01-31:
See for yourself. Spoiler: UpperFilters is UsbDk.

[ftdi.reg.txt](https://github.com/daynix/UsbDk/files/742359/ftdi.reg.txt)

dmitryfleytman, 2017-01-31:
Ok, could you please post dump of UsbView.

tsailer, 2017-01-31:
Still downloading the SDK, will take at least one more hour

tsailer, 2017-01-31:
Intel Mobile 5th Gen EHCI

tsailer, 2017-01-31:
usbview segfaults on dump (it starts up with the GUI window though)

tsailer, 2017-01-31:
[image]

dmitryfleytman, 2017-01-31:
I see, if UsbView starts with GUI, try to save log with File->Save All...
I would like to see your USB HC and Root hubs topology.

Thanks.

tsailer, 2017-02-01:
The problem is, Save from the GUI makes USBView crash too.

Without the docking station, save as text at least works, save as XML still crashes it

Without the FTDI device: [USBViewAll.txt](https://github.com/daynix/UsbDk/files/744695/USBViewAll.txt)

With the FTDI device: [USBViewAll2.txt](https://github.com/daynix/UsbDk/files/744694/USBViewAll2.txt)

dmitryfleytman, 2017-02-01:
Thanks!

Could you also provide installation log of UsbDk? You need to uninstall it with "usbdkcontroller -u" and the collect log produced during "usbdkcontroller -i"?

tsailer, 2017-02-01:
Here you go: [usbdkinstall.txt](https://github.com/daynix/UsbDk/files/744751/usbdkinstall.txt)

Uninstall did not produce any log entries

dmitryfleytman, 2017-02-01:
I don't see any UsbDk log entries there. Did you use the same UsbDk logging script?

tsailer, 2017-02-01:
No just usbdkcontroller -i. What script do I need to use?

dmitryfleytman, 2017-02-01:
I mean script for tracing, UsbDkTraceToFile.bat.

tsailer, 2017-02-01:
https://sailer.fedorapeople.org/trace1.zip

Both uninstall, then install

dmitryfleytman, 2017-02-01:
Parsed log is [here](https://gist.github.com/dmitryfleytman/2bbc960307662c0358ad43ff32ab58a9)
Thanks, I'll investigate it.

tsailer, 2017-02-01:
Similar to Issue #21 ? This is Windows 7 (x64) too.

dmitryfleytman, 2017-02-02:
Hi @tsailer ,

I did a change in UsbDk that should fix the problem you observe and prepared a [test build](https://www.dropbox.com/sh/iicsvv8opj5bkr9/AAC28nah5f2ncgD9HGwGNtdxa?dl=0).
Could you please give it a try and collect a new install and plug-in traces in case it does not work?

You will need to enable test signing mode on your system in order to load this driver, see instructions [here](https://github.com/daynix/UsbDk/blob/master/Documentation/DriverSigning.txt#L33)
Also, please uninstall original UsbDk version before installing this one.

tsailer, 2017-02-02:
Enumerate USB devices
Found 6 USB devices:
0. FilterID: 1, Port: 1, ID: 8087:8001, Configs: 1, Speed: 3
        USB\VID_8087&PID_8001 1
Descriptor for configuration #0: size 25
1. FilterID: 3, Port: 3, ID: 8087:0a2a, Configs: 1, Speed: 2
        USB\VID_8087&PID_0A2A 3
Descriptor for configuration #0: size 177
2. FilterID: 3, Port: 6, ID: 1bcf:2b8d, Configs: 1, Speed: 3
        USB\VID_1BCF&PID_2B8D 6
Descriptor for configuration #0: size 737
3. FilterID: 3, Port: 7, ID: 0a5c:5804, Configs: 1, Speed: 2
        USB\VID_0A5C&PID_5804 0123456789ABCD
Descriptor for configuration #0: size 269
4. FilterID: 5, Port: 1, ID: 0403:6010, Configs: 1, Speed: 3
        USB\VID_0403&PID_6010 AX00000003
Descriptor for configuration #0: size 55
5. FilterID: 9, Port: 1, ID: 0403:6010, Configs: 1, Speed: 2
        FTDIBUS\VID_0403+PID_6010+AX00000003B 0000
Descriptor for configuration #0: size 32

Much better, thanks !

tsailer, 2017-02-02:
There is one weirdness:

When trying to use the FTDI supplied driver with UsbDk installed, I get:
Driver file operation error. DeviceIoControl failed (The system cannot find the file specified.  Error code = 2)
But it still seems to work, after a couple of seconds delay. This output must come out of the FTDI supplied DLL

dmitryfleytman, 2017-02-02:
This printout is from UsbDk API DLL, If you collect UsbDk trace we will know what happens exactly.

dmitryfleytman, 2017-02-02:
Original issue fixed by db95b4651e079e59353d8bafa84cc5ad23c4f2b1

tsailer, 2017-02-02:
https://sailer.fedorapeople.org/trace2.zip

dmitryfleytman, 2017-02-02:
Parsed [log](https://gist.github.com/dmitryfleytman/8b220f6dd417a9b7fbd34630b592377d)

dmitryfleytman, 2017-02-02:
The problem is:

> [3]1EB8.01B8::02/02/2017-14:58:28.804 [UsbDk]CUsbDkControlDevice::AddRedirectionToSet Adding new redirection
> [3]1EB8.01B8::02/02/2017-14:58:28.804 [UsbDk]CUsbDkRedirection::Dump Redirect: DevID: **FTDIBUS\VID_0403+PID_6010+AX00000003B**, InstanceID: 0000
> [3]1EB8.01B8::02/02/2017-14:58:28.804 [UsbDk]CUsbDkControlDevice::AddRedirectionToSet failed. Cannot redirect unknown device.

Client tries to open a device by name **FTDIBUS\VID_0403+PID_6010+AX00000003B**. This name is not proper USB device name receives from UsbDk enumeration results.

Not sure why this happens, but UsbDk has nothing to do with it.

I'm closing this issue because original problem is fixed. Please open new one if there are other problems with this device.

closed 2017-02-02 by dmitryfleytman

tsailer, 2017-02-02:
It's working, just the message is irritating. So could you silence the message in the release build? Thanks.

dmitryfleytman, 2017-02-02:
This message is intended to go to the debug stream. Is it printed on the console? If so, it's client application prints it.
I would avoid silencing it because it actually indicates a real problem with client's code.

tsailer, 2017-02-02:
It ends up in the debug stream, but also on stdout. I cannot see where the client application could do that, it never does WaitForDebugEvent*

dmitryfleytman, 2017-02-02:
Let me check, maybe UsbDk prints it indeed...

tsailer, 2017-02-03:
UsbDkHelper.cpp:42

static void printExceptionString(const char *errorStr)
{
    auto tString = string2tstring(string(errorStr));
    OutputDebugString(tString.c_str());
    tcout << tString;
}

So besides outputting to the debug stream, UsbDk also outputs it via stdout.

dmitryfleytman, 2017-02-07:
Printout removed by d47bd4d2378df1c49ef064979cf8b751e09e4c5e

tsailer, 2017-02-14:
Thanks for the printout removal!

When do you plan to release an installer with that changeset included? Thanks!

dmitryfleytman, 2017-02-15:
We are going to release a new version next week after a few more fixes pushed.