daynix/UsbDk #36: Issue: USB3.0 root hub didn't work after uninstalling usbdk

xuyongjian, 2017-02-08:
Steps to reproduce:
1. UsbdkController.exe -P 0x781 -1 -1 -1 1
2. UsbdkController.exe -i (After installing, the device(vid : 0x0781) disappeared as expected)
3. UsbdkController.exe -u
After Step 3, the hub didn't work with the error "Windows cannot load the device driver for this hardware because a previous instance of the device driver is still in memory. (Code 38)"
Every time I tried to collect log, I got a blue screen error when installing usbdk.

xuyongjian, 2017-02-08:
Why can't I upload zip files? Always tell me "We don’t support that file type.  Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP."

dmitryfleytman, 2017-02-08:
Hi,

What is your UsbDk version?

~Dmitry

xuyongjian, 2017-02-08:
1.0-12

dmitryfleytman, 2017-02-08:
Please upgrade to 1.0-16 (the latest) and see if it helps. Hider API was redesigned since 1.0-12.

xuyongjian, 2017-02-08:
It really helps! Thanks Dmitry!

closed 2017-02-08 by xuyongjian