daynix/UsbDk #37: Hide rule unstable

xuyongjian, 2017-02-09:
Hi Dmitry:
With the latest code, hide rule doesn't work sometimes. The device was still loaded by the system with the correct Hardware ID.
I have sent the log to your email since I still cannot upload zip files.

dmitryfleytman, 2017-02-09:
@xuyongjian, for some reason, I cannot parse your log. My pdb just does not fit.
What is your UsbDk version and Windows version?

xuyongjian, 2017-02-10:
New log (win7-32bit, usbdk-1.0.16) sent. Check you email.
Thanks!

dmitryfleytman, 2017-02-12:
Parsed log is [here](https://gist.github.com/dmitryfleytman/0cba8dbb6020dcc9147bba7bf8350446).

From the log I see that device was reset for a few times. Could you provide exact steps to reproduce the problem?

Could you also collect UsbView [dumps](https://github.com/daynix/UsbDk/blob/master/Documentation/UsbInfo.txt) for cases when device is hidden as expected and when UsbDK has failed to hide the device?

xuyongjian, 2017-02-13:
Steps to reproduce:
1. Find a new PC that has never installed usbdk before
2. UsbDkController.exe -i
3. UsbDkController.exe -P 781 -1 -1 -1 1
4. Insert device(vid=0x0781) and it is loaded successfully(unhide).
5. Reboot the system and device still un-hidden.
6. Uninstall and reinstall the UsbDk, then add the rule, and re-plug the device. The device hides successfully.
Usbdk/usbview log was sent to you mail.

dmitryfleytman, 2017-02-13:
[UsbDkTrace-hide.txt](https://github.com/daynix/UsbDk/files/771267/UsbDkTrace-hide.txt)
[UsbDkTrace-unhide.txt](https://github.com/daynix/UsbDk/files/771268/UsbDkTrace-unhide.txt)

dmitryfleytman, 2017-02-14:
So far the log looks totally fine, not sure what is the problem. Trying to reproduce the issue locally.

Are you installing UsbDk from the official MSI?
Why do you need to run "UsbDkController.exe -i"?

dmitryfleytman, 2017-02-14:
It looks like I found a simple scenario that reproduces the problem on WIn7 x64:

* Install UsbDk
* Set hide rule for a device
* Disable and re-enable root hub corresponding to that device
* == Device becomes hidden at this stage ==
* Repeat root hub disable-enable cycle
* == Device becomes visible after 3-4 iterations ==

xuyongjian, 2017-02-15:
I once installed UsbDk from the official MSI. Then I saved the "UsbDk Runtime Library" folder so that I could install it from command line the next time, which I thought faster than MSI.

dmitryfleytman, 2017-02-19:
Hi, @xuyongjian ,

I think I fixed the problem you observe, could you, please, get [this build](https://www.dropbox.com/sh/ivu0v60ln53pnjh/AACHY-oYJvwZqNCYfLmGrSkoa?dl=0) and see if it works for you?

You'll need to [enable test signing](https://github.com/daynix/UsbDk/blob/master/Documentation/DriverSigning.txt#L33) on your machine and uninstall any other UsbDk MSI packages before installing this one.

Thanks,
Dmitry

xuyongjian, 2017-02-21:
Hi @dmitryfleytman 
I tested the build on win7-32bit and win10-32bit for a few times, and glad to see that the problem solved!
Thanks very much!

closed 2017-02-21 by xuyongjian

dmitryfleytman, 2017-02-21:
Fixed by 6f383cb2b90a83f8b8764776b1c57cbceb44a3db