daynix/UsbDk #40: usdbk device prevent device usage in virtual-box

diabolo38, 2017-03-02:
Under win7-64 (home edition)
After usb device  (maybe) used  by  usbdk  via libusb and released it may not be possible
anymore to map it in a virtual machine( virtual biox) 
I'm not 100% how i reach that situation,  i noticed it after a while installing playing and  testing libusbd+usbdk  i was not able  anymore to use an device  in a  VirtualBox  , it  fail to  acquire device in host and map it on  the guest. 

I had to play with registry editor and delete an entry "upfilter" that  was set as "Usdbk"  in order to get back usb device usable in virtual machine. 
I came to this solution described in many virtual box forum post like here https://forums.virtualbox.org/viewtopic.php?f=2&t=37326

Is this a know issue? It's  quite annoying for  libusb based app under window if the same device is to be used in guest  os (typically my multi os test environment).

dmitryfleytman, 2017-03-13:
Hi,

Thanks for the report. Looks like interoperability issue between UsbDk and Virtual Box drivers.
We will check this. It would be great if your could provide exact steps to reproduce this problem.

Thanks,
Dmitry

diabolo38, 2017-03-19:
From a fresh window say "no usbdk" (de-installed+regisrty clan manualy already) i can  attach my test dev (stm32 cdc/acm) to virtual box guest os.
After usbdk runtime get install from https://www.spice-space.org/download/windows/usbdk/UsbDk_1.0.17_x64.msi
and re-started window I can't attached my device to virtual box guest , 
The registry key mentioned is set again and seam to cause issue. 
So their is not even  need to use the device with libusb  to cause the issue installing usbdk is enough to cause interoperability issue with  virtual box.

Strange things i noticed  : while de-installing and same while installing "usbdk"  caused my VIA USB 3.0 root hub to appear with exclamation mark , also running virtual box  stop to show many usb-device that were presents before install/de-install.  the mark goes away after os reboot but never usbdk  asked to restart during (de-)install?

dmitryfleytman, 2017-03-20:
Thanks, @diabolo38 

We will check this.

eric-schleicher, 2017-05-23:
I can confirm this behavior on windows 10 x64.  if usbDK is installed, cannot attach USB devices to VirtualBox Guest.

sameehj, 2017-07-12:
A patch which fixes the issue was sent to Vbox-devel:
https://www.virtualbox.org/pipermail/vbox-dev/2017-July/014625.html
For more info on what causes the issue you can read the cover letter:
https://www.virtualbox.org/pipermail/vbox-dev/2017-July/014624.html

closed 2017-07-12 by sameehj

mheyer32, 2024-12-20:
I just want to say that this is STILL an issue. I just ran into it with VirtualBox 7.0.22r165102

gunjanjambhulkar, 2026-09-04:
I encountered what appears to be the same issue on a newer Windows/UsbDk/VirtualBox setup.

Environment:

UsbDk 1.0.22 x64
UsbDk installed as part of Analog Devices CCES 3.1
VirtualBox 7.1.4 and later 7.2.2
UpperFilters under {36FC9E60-C465-11CF-8056-444553540000} contained UsbDk

With UsbDk registered in UpperFilters, VirtualBox was unable to attach multiple unrelated physical USB devices. I tested, among others:

Silicon Labs CP2102N USB-UART
Jabra USB devices
Saleae Logic Pro
Analog Devices USB device

The typical VirtualBox error was:

USB device is busy with a previous request

E_INVALIDARG (0x80070057)

The problem occurred with both VirtualBox 7.1.4 and 7.2.2.

After removing the UsbDk entry from UpperFilters and rebooting Windows, VirtualBox USB passthrough started working again. The CP2102N and Analog Devices USB device could then be attached to the VM successfully.

I also observed another potentially related behavior: when I temporarily configured the UsbDk driver not to start while UpperFilters=UsbDk was still present, USB keyboard and mouse stopped working after reboot. Restoring the UsbDk driver startup configuration and rebooting restored USB input.

This latter behavior may also be related to #134.