daynix/UsbDk #41: failed to install on Windows 10 and USB doesn't work

liugengdai, 2017-03-09:
I tried to install usbdk on my windows 10 desktop. The installation was blocked and did not succeed. Now all my USB ports don't work. I cannot enter windows now. I think I should find a bluetooth or wire keyboard and mouse. :(

Do you know what happened and how to solve the problem?

dmitryfleytman, 2017-03-10:
Hi,

You have to boot in safe mode and uninstall UsbDk.

Are you using Windows10 RS1 with secure boot enabled?

~Dmitry

dmitryfleytman, 2017-03-13:
Most likely the same as #42

theiman2, 2017-04-17:
How do I uninstall it in safe mode? Running the .msi again just attempts to install, not uninstall.

dmitryfleytman, 2017-04-18:
@theiman2, either go to control panel/uninstall software are remove UsbDk Runtime Libraries or navigate with command line to c:\program files\UsbDk runtime Libraries and run "UsbDkController.exe -u"

dmitryfleytman, 2017-04-20:
Fixed in v1.00-19

closed 2017-04-20 by dmitryfleytman

dmitryfleytman, 2017-05-25:
Documentation for uninstallation of UsbDk after signing error may be found [here](https://github.com/daynix/UsbDk/blob/master/Documentation/Uninstallation.txt#L15).