daynix/UsbDk #42: Installation o Windows 10 blocks all USB

zeroxm, 2017-03-11:
Tried to install on Windows 10, as recommended in this emulation project,  https://wiki.dolphin-emu.org/index.php?title=USB_Passthrough#In_5.0-2352_and_newer

I know it's not the first time this is reported, but I you should consider marking the lib linked above to be incompatible with Windows 10.

I'll try to get a PS/2 keyboard and Uninstall the driver tomorrow. I'll try and report back and give some more info about it. Currently I can say:
- Windows says it's an unsigned driver,  maybe this is what's causing this behavior
- There is a way to test whether adding permission to unsignes drivers makes any difference,I'll try to do it

zeroxm, 2017-03-11:
Confirmed.

If you change the boot options so it disable driver signature enforcement the simptoms disapear (currently I still can't boot without it)

dmitryfleytman, 2017-03-13:
Hi,

You should be able to boot in safe mode and uninstall UsbDk.

Are you using Windows10 RS1 with secure boot enabled?

~Dmitry

zeroxm, 2017-03-13:
I was on Windows 10,  and yes I was able to do so,  but turns out just uninstalling it using Control Panel doesn't fix this, it looks like the Driver were still there.

I tried something stupid after and thought I could manually delete the Driver, and Windows try and fix this.... but no.

dmitryfleytman, 2017-03-13:
You need to uninstall the driver and then reboot the system.

zeroxm, 2017-03-13:
Oh and Yes, secure boot didn't let me mess with driver signature checking.

In the end I had to reinstall windows  (formatting it) because I needed the PC to work again.

dmitryfleytman, 2017-03-13:
Sorry to hear that, we are working on solution that will include full signature suitable for Win10 RS1 with secure boot and verification of driver functionality during installation.

zeroxm, 2017-03-13:
I did Unninstall and reboot, but the USB wouldn't work unless I did something similar to Solution 1, shown here.

http://windowsreport.com/driver-signature-enforcement-windows-10/

zeroxm, 2017-03-13:
Anyway I hope the info provided helps, if you need something else, please, do ask.

Waishon, 2017-03-31:
Is there a new status about this issue?

MarcusWichelmann, 2017-03-31:
Are you still working on this?
We planned to recommend the use of this driver to pupil of our school to be able to connect to our spice VMs, but since it just killed my Windows10 PC and I've read this thread we probably will not for near future.

I've been able to recover the USB support of my PC by logging in via RDP and uninstalling the driver in device-manager.

dmitryfleytman, 2017-04-02:
Hi @MarcusWichelmann,

Yes, we are working on this. We expect that the first public release with all signatures and full Windows 10 support will be public in 2-3 weeks.

Thanks for using UsbDk,
Dmitry.

dmitryfleytman, 2017-04-20:
Fixed in v1.00-19

closed 2017-04-20 by dmitryfleytman

matrix65, 2017-05-09:
With v1.00 - 19, on windows 10 x64,  version 1703, 15063.138 build, the driver is still blocked. 
Running cmd(admin) UsbDkController.exe -i  only got "UsbDk driver installation failed"

zeroxm, 2017-05-09:
I've tested the driver on v1.00-19 and it worked as intended. 
Maybe something more specific on your installation?

MarcusWichelmann, 2017-05-09:
The driver works as expected here, now. I can confirm, that this issue is fixed

dmitryfleytman, 2017-05-10:
Hi @matrix65 , are you installing officially released binaries or build the driver yourself?

In case you're using official binaries, could you please collect [user mode logs](https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L15) to see what's wrong with the installation?

matrix65, 2017-05-11:
It turns out that it was the external usb drive that blocked the installation. After removing it and a clean reboot, the driver is sucessfully installed. 
Thanks for all the above quick responses to help me.