daynix/UsbDk #45: Provide a single 32bits + 64bits MSI package

Ecco, 2017-04-22:
Sure it would be bigger (and a waste of space in most cases), but it would be so much easier to install for end users. Most don't have a clue wether they're on 32 or 64 bits systems.

dmitryfleytman, 2017-04-23:
Sounds like a good idea. I'm adding this to project task list, however not sure we will get to this soon.

Any volunteers for this task?

ybendito, 2017-07-16:
There is limitation of MSI: I can't be x86 and x64 at the same time. When there are possible workaround's directions, this is not trivial and time consuming to combine both installations into single MSI.
Discussion on this matter with some additional links:
https://social.msdn.microsoft.com/Forums/windows/en-US/203adade-74d8-4d6f-b238-f4af23c2d558/single-installer-for-32bit-and-64bit-machines?forum=winformssetup

closed 2017-07-16 by dmitryfleytman