daynix/UsbDk #46: UsbDk don't work ok when redirect a camera from a Windows XP

suqixiong, 2017-07-25:
Hi,

When I redirect a camera from a Windows XP SP3 into a virtual machine by UsbDk，the amcap in the virtual machine always show balck screen and there are no images displayed.

The version of usbdk is 1.0.19 and the etl file and USB Information of the camera is in the attachment. USBView can't dump in Windows XP, so I used USBTreeView instead.
[UsbDkTrace.zip](https://github.com/daynix/UsbDk/files/1171865/UsbDkTrace.zip)

Do you know the reason why this happens?

Thanks,
Qixiong

ybendito, 2017-07-26:
Hi Qixiong,
Several questions:
1. In the ETL file we can see you redirect camera device USB\Vid_1908&Pid_2310, when in UsbView files the device attached is USB\Vid)093a&Pid_2700; possible due to the fact that ETL record was taken 3 weeks earlier than the UsbView. Can you please sync the files, i.e. make ETL record with actually attached device?
2. Which OS you use on VM?
3. Which client software you use on XP machine? If I'm not mistaken, the 'remote viewer' that supports XP does not support USB redirection.
4. Did you have a chance to redirect this camera device camera from Windows client other than XP?
5. Does redirection of other camera devices work on your XP client?
Thanks

suqixiong, 2017-08-02:
Hi ybendito,
Thanks for your reply.

1. New log files is in the attachment.
[UsbDkTrace.zip](https://github.com/daynix/UsbDk/files/1192531/UsbDkTrace.zip)
2. The OS of VM is win7 64bit.
3. I am using the spicy client compiled from spice-gtk.
4. If the OS of Client Machine is win7 or win10, the camera works ok on VM;
5. Other camera devices also doesn't work on my XP client.

ybendito, 2017-08-17:
Hi suqixiong,
Can you share with us spice client that we can use on XP 32/64?

Thanks

day1212, 2019-05-27:
Hi ybendito：
 i have find the same issue when redirect camera from win7，the amcap in the virtual machine always show black screen.The return status of Isochronous in transfer is 0, but data length 12.

spice client is remote-viewer 6.0
version of spice-gtk is 0.33
version of spice protocol is 0.14
version of libusb is 1.0.20
version of usbDk is 1.0.21