daynix/UsbDk #47: "A driver without a digital signature" warning when UsbDk install on Windows 10

jik2017, 2017-07-27:
When UsbDk install on Windows 10,  system will wairng that  "A driver without a digital signature" ,
but the UsbDk driver is really be signed.  WHQL is required?

os version: 10.0.15063

dmitryfleytman, 2017-07-27:
Hi,

The WHQL is not required for this type of drivers.
Where do you see this warning? Could you post screenshot?

~Dmitry

nikola-matic, 2017-08-19:
@dmitryfleytman 

Just to tag on, I'm also getting a similar (or perhaps the same) error during a Windows 10 install, using the MSI installer.

[image]

In addition,  when I try running UsbDkController.exe, all I get is "UsbDk driver installation failed", so no specifics regarding the exact cause.

OS: Win 10 - Version: 1703, Build: 15063.540

dmitryfleytman, 2017-08-20:
@nikola-matic ,
Are you using official UsbDK build or UsbDk compiled manually?
Which version of UsbDk are you using?

nikola-matic, 2017-08-20:
@dmitryfleytman, I'm using the compiled version; where can I find the official build?

dmitryfleytman, 2017-08-20:
Releases can be found at https://github.com/daynix/UsbDk/releases

If you compile UsbDk yourself, you need to sign it with your key.
Official releases are signed with RedHat key during build.

closed 2017-08-20 by dmitryfleytman

nikola-matic, 2017-08-20:
@dmitryfleytman Fantastic, thanks.

jelabarre59, 2020-04-22:
Vanilla MSWindows 7 install still sees v1.00-22 as unsigned.  This is a copy downloaded from here, not one I have built.