daynix/UsbDk #50: BSOD on Windows 10

Ecco, 2017-09-14:
We're using UsbDk to talk to a DFU device on Windows.

Some of our users encounter a BSOD whenever they plug our device to their machine.

The screen says
```
Stop Code: DRIVER IRQL NOT LESS OR EQUAL
What failed: UsbDk.sys
```

Version v1.00-19

sameehj, 2017-09-17:
@Ecco Hi,

In order to debug the issue efficiently I need you to provide me with more information on the crash. On which OS version this crash occurs? Is the source code that you are using is publicly available? Do you know which part of your code leads to UsbDk crashing? Do you know how to reproduce this on any instance running your software? If so then please provide me with the info and instruction so that we can investigate this further.

If you can provide me with the dump file that was created by Windows after the crash, it would be great as it has most of the information I need to investigate the crash. The dump file usually resides in "c:\windows" and it is called "memory.dmp".

sameehj, 2017-09-17:
Please also check this manual on dump files:

https://github.com/daynix/UsbDk/blob/master/Documentation/CrashDumps.txt

sameehj, 2017-10-18:
Hi @Ecco,

Since you haven't reply for a while now I am going to close this issue for now,
please feel free to reopen if this is still relevant.

closed 2017-10-18 by sameehj