daynix/UsbDk #51: USBs don't work after uninstall UsbDk v1.00-17

adri1992, 2017-10-15:
Hi everyone!

I've uninstalled the UsbDk v1.00-17 and the usb ports are not working now on Windows 10. Windows is installed with Ubuntu, through the GRUB, and I tried to enter safe mode after select Windows on the Grub menu and, because the usb ports, I can't enter on the safe mode... 
After I select Windows 10, all usb stops and it is impossible to make any action.

On Ubuntu all is working correctly.

Any idea of how can I solve this?

sameehj, 2017-10-18:
Hi @adri1992,

You can find relevant info on how to get into Safe Mode using the following documentation:
https://github.com/daynix/UsbDk/blob/master/Documentation/Uninstallation.txt#L35
In general after two failed starts, Windows automatically enters to the Recovery menu where from there you can enter the Safe mode.

It is important that you use the  v1.00-19 of UsbDk as this issue was already fixed in there. 

Please update me on your progress :)

adri1992, 2017-10-30:
I have no idea of what action has solved the problem, but after entering into "Repair mode" of windows 10 the problem disappeared without any executed action =/
Very strange... 

This issue can be closed.

Thanks for the help!

sameehj, 2017-11-04:
Hi @adri1992,

Thanks for the feedback, just out of curiosity, what do you mean by "repair mode" in Windows 10? I believe that whatever mode you are referring to it had "Driver signature enforcement" disabled as this is the root cause of the issue you described.

closed 2017-12-11 by sameehj