daynix/UsbDk #53: v1.00-19 fails to install on Windows 7

matlo, 2017-12-11:
This was reported by a GIMX user here:
https://gimx.fr/forum/viewtopic.php?p=12479#p12479

This is on Windows 7 SP1 x64.

matlo, 2017-12-11:
Note that v1.00-15 worked fine.

sameehj, 2017-12-11:
@matlo Hi,

In order to debug this effectively, can you please provide me with some traces? And if possible share to the dump file it would be great!

You can find information on how to collect these in the following:
https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt
https://github.com/daynix/UsbDk/blob/master/Documentation/CrashDumps.txt

sameehj, 2017-12-11:
> Will there be a crash dump if installation fails?
Nope, only if a BSOD occurs.

matlo, 2017-12-11:
Is there any trace or debug information to retrieve during installation?

sameehj, 2017-12-11:
> Is there any trace or debug information to retrieve during installation?

Yes, please refer to the tracing document that I referred you to in earlier comment:
https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt

If you need help or something is not clear with the instructions please update me so we can resolve this =)

matlo, 2017-12-11:
I'm unsure about which section(s) is(are) relevant for the issue.

sameehj, 2017-12-11:
Update the user to follow these instructions:

* Copy Tools/Trace subfolder from the driver source tree to the target machine
* Run UsbDkLogman.bat as administrator
* Follow instructions

sameehj, 2018-01-15:
Any updates on the matter?

matlo, 2018-01-15:
Sadly not. I will reopen the issue it if user provides the requested information.

closed 2018-01-15 by matlo