daynix/UsbDk #54: BSoD when the installation is almost complete

CrashFinder, 2018-02-08:
Hello I am a new user and this is my first post.
Just a quick recap before explainind the issue; In december I found a BSoD during the update of a software called GIMX and it were reported on the official forum here: [http://gimx.fr/forum/viewtopic.php?f=5&t=2370&start=10](url)
Matlo, the author of GIMX, had opened a ticked on Github here [https://github.com/daynix/UsbDk/issues/52](url). Has I explained I successfully found a way to resolve the issue on my windows 7 laptop and the ticket were closed.

In the attempt to replicate che original issue I have re-done all the steps explained in the original ticket. I successfully replicate the crash but windows failed to create the dump. So I tried to resolve the issue and complete the installation of the GIMX software by replicating the steps I found described into the original ticket. Unfortunately they don't work anymore; each attempt to install USBDK (during GIMX installation) causes that windows goes into BSoD. Then I tried to install USBDK in DOS but it fails without explanation.
Actually I'm stuck and I cannot install USBDK nor use the GIMX software.
I have uploaded each dump created after each installation attempts and other files here [https://drive.google.com/drive/folders/1hufIb-Ar--jtoC6gLLC_KgufPl66Qb3J?usp=sharing](url)

Also, I have been notified with a strange dialog; if I launch an update/install of another software (virtualbox in my case) windows notify that (check image files "suspended USBDK install") another installation is **suspended**; the dialog refers to the USBDK installation which has been interrupted by the BSoD; it asks the user that a confirm is needed to proceed with the installation of (virtualbox) and explains that the confirm will void the changes made by the previous installation process.
Please help me!
THXS

YanVugenfirer, 2018-02-12:
Thank you for reporting the issue. 

Can you please explain what you mean by "Then I tried to install USBDK in DOS"? Thanks.

We will look at the dump files to try to figure out what is happening.

CrashFinder, 2018-02-12:
Hello,
I mean by windows command prompt; Isn't a DOS console which runs into windows OS? Just asking because I'm not a programmer and I don't want to say something while meaning another.
Just an update about the behavior of the GIMX. When the install is almost complete  the bluetooth controller is everytime turned on automatically (a dialog appears and a pop-up is notified on the windows bar) and after some seconds the PC goes in BSoD.
I have checked bluetooth peripherals and deleted all of them. Then I uninstalled USBDK by command prompt and the GIMX software fro mthe control panel; re-launched GIMX installer and it ended correctly. Now everything works fine.
I don't understand if there is a bug (which causes the BSoD) with USBKD install process if a peripheral (like my Logitech G27) is connected to the PC or there is some conflict with a bluetooth device. Hope that the dumps will help you and answer my question.

ybendito, 2018-02-12:
From dump analysis I see the crash is originated by hcmon.sys driver which I think is related to VirtualBox. Is VirtualBox installed or your machine? Is it possible to try the installation of UsbDk/GIMX without VirtualBox?

CrashFinder, 2018-02-12:
Yes, I have virtual box but I don't use it with GIMX and it was not running during the install/uninstall attempts.
I can help you in testing the install process without virtualbox but keep in mind that something is changed because  I have already installed gimx+USBDK successfully after having deleted the bluetooth peripherals.
What should I do?
Deleting gimx and usbdk and virtual box and re-install gimx+usbdk
or
uninstall virtualbox+gimx+usbdk, then install an old version of gimx and trying to replicate the initial BSoD by updating the gimx to a newer version while my G27 wheel is still connected?
Tell me which test is better for you.

PS:
yes, I have notice something related to hcmon.exe... I tried to repair it and found the "suspended installation dialog" already described :)

ybendito, 2018-02-12:
When VirtualBox is just installed and not running, it still has hcmon.sys driver connected to all USB devices (UsbDk is also, BTW). It is possible that BT device triggers some problematic flow in hcmon when UsbDk present. Both options you suggested are good.

CrashFinder, 2018-02-12:
OK, I will perform both test. Give me some time. I will update asap.

CrashFinder, 2018-02-13:
First update;
I already had two crashes; I did:
uninstalled virtualbox, then gimx (sorry not remember the order) and finally usbdk. Then I checked usbdk folder and noticed that files were still there (is correct that some files remain there???); so I uninstalled usbdk via command prompt. Then a crash happened but windows fails to create the dump (bsod didn't appear).
After the reboot I entered into safe mode and completed the uninstall of usbdk via command prompt.
Reboot in normal mode
Installed gimx 6.11 (not the final version) and another crash happened (as usually when the installation is almost finished). The bsod screen refers again to hcmon.sys. I'm still uploading the dump; it will take two hours from now.
**Virtualbox has not been installed yet, so why is there a sort of relationship with it?**
In addiction I have checked usb peripherals with USBView tool; something wrong is reported. I saved the report in xml and as image (usb all. gif). They have already been uploaded. Please take a look.
Now, should I install virtualbox or do you prefer to do some test based on the new dump?

ybendito, 2018-02-13:
Can you please provide a link to latest dump file after VirtualBox is uninstalled? (the link in comment 1 is not valid). I'd expect that hcmon driver is uninstalled together with VirtualBox. If it still present, I'd even suggest to delete the service. You should be able to check it with "sc query hcmon", and stop/delete with "sc stop hcmon" and "sc delete hcmon".

CrashFinder, 2018-02-13:
I have shared the entire folder so you should have access to each file. Doesn't the share work?
Anyway here the link to the dump
https://drive.google.com/open?id=1TOw4qy89iMNZvrzRwP7h3K7MCTGYFzIK

folder link
https://drive.google.com/open?id=1hufIb-Ar--jtoC6gLLC_KgufPl66Qb3J

I have opened command console and typed "sc query hcmon" -> the service status is "4 running" (stoppable     not_pausable    ignores_shutdown)
then typed "sc stop hcmon" -> status changed to "3 stop_pending"
then typed "sc delete hcmon" -> deleteservice successfully
Now I think I could try to reinstall gimx+usbdk, right?

ybendito, 2018-02-13:
I would restart the system and verify that the hcmon does not exist as the service anymore

CrashFinder, 2018-02-18:
The service does not exist anymore.
Next, I installed gimx 6.11+usbdk in a normal windows session through the installer. Everything gone fine. Then I restarted the pc and installed virtual box. The service hcmon still doesn't exist.
Next, I used gimx for abount 1 hour then I updated it to 7.2 (the newer version) while the G27 was still connect. Updated install gone fine. So everything seem ok now. I will do other test during the week. Do you have test suggestions?
PS: why the hcmon service doesn't appear anymore?

matlo, 2018-02-19:
I think hcmon is related to VMware, not VirtualBox.

YanVugenfirer, 2018-02-19:
@matlo You are correct. This is VMWare USB monitor.

YanVugenfirer, 2018-02-19:
> PS: why the hcmon service doesn't appear anymore?

It could be a remnant of some old VMWare (VMWare Player?) installation.

YanVugenfirer, 2018-02-19:
I think we can close this issue for now.

closed 2018-02-19 by YanVugenfirer

matlo, 2018-02-19:
@CrashFinder
Any idea about the VMware version you installed earlier? Could you try installing the latest version and see if this still happens?

CrashFinder, 2018-02-24:
@matlo 
No, I'm sorry.. it was many years ago; I installed it because there was a conflict between virtualbox and avira antivirus which taken several months to be fixed.
Anyway I have downloaded the latest version of vmware and now the service hcmon is again here. I will then perform again all the steps. Give me some time :)

CrashFinder, 2018-03-02:
@matlo
Hi Matlo, I have some news. After the installation of WMware, I have launched the GIMX 7.1 install which crashed during the finalization of the installation. It caused a BSOD which is related to the hcmon service (I have the dump if you need it).
Then I tried to stop and delete the hcmon service but despite the confirmation I had in the command prompt, the service was still there and the pc crashed again when I tried to install the gimx again.
The next reboot I deleted th WMware application. Stopped and deleted the hcmon service in the command prompt and then installed the gimx again. The installation was successfully completed (I haven't checked if gimx works but it seems ok).

matlo, 2018-03-07:
@CrashFinder 
Which WMware version did you install? Could you please provide a link?

CrashFinder, 2018-03-07:
For the latest test I installed the latest version of WMware which has been downloaded from the official site https://www.vmware.com/it.html - download section. The package name is VMware-workstation-full-14.1.1-7528167.exe

matlo, 2018-03-11:
I think this issue should be reported to VMWare. Maybe here: https://communities.vmware.com/community/vmtn/workstation/content?filterID=contentstatus[published]~objecttype~objecttype[thread]

@YanVugenfirer maybe daynix / Red Hat could help reporting this problem to VMWare?