daynix/UsbDk #55: external USB HUB: Redirect of USB device failed

Manel3, 2018-02-11:
First of all thanks for your great work!

I am trying to access from Windows a new D-Link DUB-H4 H/W Ver.:D1 05e3:0608 because it has power-per-port control from uhubctl. That already works on Linux and you can really switch off +5V of every port.

It does not work on Windows, and using UsbDkController.exe -r I see the problem is on usbdk, below you have enumeration, redirect of the D-LINK (failing) and redirect of an optical mouse (which do work):

C:\USB-Win>"c:\Program Files\UsbDk Runtime Library\UsbDkController.exe" -n
Enumerate USB devices
Found 4 USB devices:
0. FilterID: 1, Port: 1, ID: 05e3:0608, Configs: 1, Speed: 3
        USB\VID_05E3&PID_0608 1
Descriptor for configuration #0: size 25
1. FilterID: 1, Port: 5, ID: 8087:0aa7, Configs: 1, Speed: 2
        USB\VID_8087&PID_0AA7 5
Descriptor for configuration #0: size 177
2. FilterID: 1, Port: 7, ID: 0bda:57b3, Configs: 1, Speed: 3
        USB\VID_0BDA&PID_57B3 200901010001
Descriptor for configuration #0: size 373
3. FilterID: 1, Port: 2, ID: 046d:c05a, Configs: 1, Speed: 2
        USB\VID_046D&PID_C05A 2
Descriptor for configuration #0: size 34

C:\USB-Win>"c:\Program Files\UsbDk Runtime Library\UsbDkController.exe" -r "USB\VID_05E3&PID_0608" 1
Redirect USB device USB\VID_05E3&PID_0608, 1
Redirect of USB device failed

C:\USB-Win>"c:\Program Files\UsbDk Runtime Library\UsbDkController.exe" -r "USB\VID_046D&PID_C05A" 2
Redirect USB device USB\VID_046D&PID_C05A, 2
USB device was redirected successfully. Redirected device handle = 000001DF8352BF70
Press any key to stop redirection
Restore USB device 000001DF8352BF70
USB device redirection was stopped successfully.

If redirection is not going to work, could I use usbdk "layer" to send a control transfer to the hub without redirecting it (keeping it working for windows)?
That is what seems to be really needed to check and use power-per-port control, and I do not have that clear after reading all your doc.

ybendito, 2018-02-13:
If I understand correctly, the device you're trying to redirect is the HUB, when UsbDk is intended to redirect  USB devices only, not hubs. The interface that UsbDk provides (to libusb, for example) also allows access only to redirected devices, not to hubs.

Manel3, 2018-02-14:
Yes I want to "talk" to the HUB, and I have seen libusb starts redirecting the USB device when opening it.
Could I try to use usbdk interface without redirecting the USB HUB? I just need some control transfers.
Any tip for that?

ybendito, 2018-02-14:
Unfortunately, the UsbDk does not support redirection of HUB devices even if libusb requests it. I guess that even if you modify UsbDk to allow redirection of HUB devices, immediately upon redirection the HUB will stop functioning as a HUB and will appear as raw device without hub capabilities.

mvp, 2018-02-16:
@ybendito: as far as I can tell, if someone wants to turn power off on some or all ports of given hub, they probably don't care that much if hub stops working as a hub. All they care about is having power turned off on hub port, and once power turned off (by sending low level control transfer message to hub hardware chip), usbdk redirection can be immediately stopped.

However, I don't know if usbdk is actually implementing it's own kernel USB driver, or if it's simply using winusb.sys for all low level USB access. If winusb.sys is real workhorse here, then answer to @Manel3 question would be no.

Manel3, 2018-02-16:
I took a look at the code. I have found a call to WdfUsbTargetDeviceFormatRequestForControlTransfer (see https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/working-with-usb-devices#a-href-idsending-a-control-transfera-sending-a-control-transfer) so a direct control transfer without winusb is done but I would say only for redirected devices (Am I right?) and never for hubs.

The idea of a filter driver patching winusb.sys (really usbhub3.sys) (or any other driver) for all hubs is very good, but I think it can be outside the scope of usbdk which was really designed to redirect USB devices for spice protocol for remote virtual machines.

ybendito, 2018-02-19:
That's correct, UsbDk allows sending commands only to redirected devices and it does not redirect hubs.

closed 2018-02-19 by YanVugenfirer