daynix/UsbDk #56: UsbDk_StopRedirect　need open UsbDk device link?

goldendeng, 2018-04-11:
I have a question about this code:[UsbDk_StopRedirect](https://github.com/daynix/UsbDk/blob/master/UsbDkHelper/UsbDkHelper.cpp#L176)

In fact, it can stop redirection without using this operation.

YanVugenfirer, 2018-04-11:
HI,

Can you please reconfigure https://github.com/daynix/UsbDk/blob/master/.appveyor.yml in your repo to send notifications to you instead of devel@daynix.com? 

Thanks.

YanVugenfirer, 2018-04-11:
Sorry, I didn't understand the question. What do you refer to as "it"?

Thanks.

goldendeng, 2018-04-11:
I'm sorry about this, but I have changed it now.

goldendeng, 2018-04-11:
I mean, no such code, this function can run normally, so I want to ask about the function of this code.

goldendeng, 2018-04-11:
@sameehj  Yes, I think so.

sameehj, 2018-04-11:
You are right, can you please send a pull request that resolves this?

ybendito, 2018-10-09:
Note that this is not a 'local variable'. This is an object with constructor that raises exception in case it can't access the driver from the helper at the moment of the call. It also creates additional handle on the driver's device file which is closed when the UsbDk_StopRedirect procedure is fully completed. Such referencing is typical to protect the flow from corner cases. I would revert the commit until there is more serious reason to remove this reference than just 'Stop redirect works without it'.

YanVugenfirer, 2018-10-09:
I would revert the commit and rename the variable to reflect the actual functionality that happens during its creation.

PR #58 (2018-04-11): Delete unused local variable
Closes #56
commit f2dae220 (2018-04-11): Delete unused local variable

Closes #56
closed 2018-10-14 by ybendito