daynix/UsbDk #60: Crash using Logitech Vantage Mic (Rock Band) with USB Passthrough Feature (Isochronous Transfer) of Dolphin Emulator

PipeRojas, 2018-05-29:
Hi everybody, I'm having an issue, I'm trying to play guitar hero and rock band on dolphin emulator with the USB Logitech vantage microphone with the USB passthrough feature and usbdk 1.00-19 but after registering the device in the white list of dolphin usb passthrough and starting the game, it crashes randomly, sometimes I can play a half of a song and dolphin crashes with Windows 10 Pro x64 up to date (fall creators update).
Anybody knows any solution?

And thanks a lot btw

PipeRojas, 2018-05-29:
https://1drv.ms/u/s!Amvh0udatWAwiFw54anpi7CadZ1J

Here´s a dump file of Dolphin just before it crashes.

Tnx for the help

InlineData, 2019-02-06:
Hello,

I have been having similar issues with the Logitech Vantage Mic with Dolphin. Talking to the Dolphin community we believe that a bug originates from UsbDk. If you are interested the discussion can be found at: https://bugs.dolphin-emu.org/issues/11511

It would be really great if this issue with Isochronous Transfers could be resolved so that Rock Band on Dolphin would be fully playable. Any help would be greatly appreciated. Some relevant information that might help:

Call Stack:
```
    [Inline Frame] Dolphin.exe!list_del(list_head *) Line 137   C
    [Inline Frame] Dolphin.exe!usbi_disconnect_device(libusb_device *) Line 749 C
    Dolphin.exe!libusb_unref_device(libusb_device * dev) Line 1189  C

    Dolphin.exe!IOS::HLE::Device::USBHost::AddNewDevices(std::set,std::allocator > & new_devices, std::mapstd::shared_ptr<IOS::HLE::USB::Device,enum IOS::HLE::Device::USBHost::ChangeEvent,std::lessstd::shared_ptr<IOS::HLE::USB::Device >,std::allocatorstd::pair<std::shared_ptr<IOS::HLE::USB::Device const ,enum IOS::HLE::Device::USBHost::ChangeEvent> > > & hooks, bool always_add_hooks) Line 150 C++
    Dolphin.exe!IOS::HLE::Device::USBHost::UpdateDevices(bool always_add_hooks) Line 116 C++
    Dolphin.exe!IOS::HLE::Device::USBHost::Open(const IOS::HLE::OpenRequest & request) Line 50 C++
    Dolphin.exe!IOS::HLE::Device::OH0::Open(const IOS::HLE::OpenRequest & request) Line 39 C++
    Dolphin.exe!IOS::HLE::Kernel::OpenDevice(IOS::HLE::OpenRequest & request) Line 507 C++
    Dolphin.exe!IOS::HLE::Kernel::HandleIPCCommand(const IOS::HLE::Request & request) Line 523 C++
    [Inline Frame] Dolphin.exe!IOS::HLE::Kernel::ExecuteIPCCommand(unsigned int) Line 574 C++
    Dolphin.exe!IOS::HLE::Kernel::UpdateIPC() Line 639 C++
    Dolphin.exe!CoreTiming::Advance() Line 327 C++
    [External Code]
```

Screenshots:
[image]
[image]

Thank you for your help.

ybendito, 2019-02-06:
Hi all,
We are maintainers of UsbDk which is used by LibUsb as backend providing access to local USB device. UsbDk is widely (mass production) used with LibUsb in Spice/Virt-Viewer packages and we did not face problems like one described for years.
We were not in touch with developers of 'Dolphin' application; reviewing the dump file provided by @PipeRojas we do not see UsbDk in the call stack (UsbDkHelper.dll is loaded in the process), so I do not see any possibility that the application crash is _originated_ by UsbDk. In the crash analysis by @InlineData the crash happens when the application tries to unreference libusb device object, so it is possible that it tries to unreference the device one more time than needed. What definitely can help here is detailed log of libusb. For that it shall be initialized with 'debug' flag, then it will emit its log entries (the application shall collect them, if I remember correctly the LibUsb emit logs to stderr, so the application shall redirect it). LibUsb is open source and can be easily built using Visual Studio or mingw with any additional debug information. In role of maintainer of the application I would definitely start from this.
You can also try to use WinUSB instead of UsbDk (if I'm not mistaken, this backend is also supported by LibUsb).

InlineData, 2019-02-07:
Hello,

Preamble and context:
Just to clarify that I wouldn't consider myself a developer for Dolphin. I haven't made any contributions to the code base. I would consider myself just an enthusiast who contributes issues to the Dolphin issue tracker sometimes. I came here on my own to try and resolve and issue which may be a part of Dolphin or UsbDk to fix a problem in Rock Band.

Thank you for getting back to me to soon. It could be very well that Dolphin unreferences libusb more than once. Personally, I haven't contributed any code to Dolphin myself so I couldn't be sure of such an occurrence couldn't be happening. Apparently the Linux version of Dolphin doesn't have such issues though using the Logitech Vantage Mic with usb passthrough. I don't know if that would rule out issues on Dolphin's side though. I agree that we cannot be sure without a log from libusb. I will try to get that to you as soon as I have the development environment ready and have a chance to do testing.

I attempted to install the WinUSB  (v6.1.7600.16385) device driver with Zadig 2.4 on the Logitech Vantage Mic. Testing on Dolphin 5.0-9277 for usb passthrough with that device with the modified device drive rendered no success. The device wasn't recognized in Rock Band. No crash though. Thank you for the suggestion. I believe that is what you meant by using WinUSB instead of UsbDk unless I am mistaken.

Thank you so much for you time on this matter.

ybendito, 2019-02-07:
Using LibUsb on Linux and on Windows is not exactly the same. The main difference is processing of device arrival/removal (at least LibUsb on Linux supports hot plug indication, on Windows does not, so the application shall work differently, there are also more differences). On Windows, at the moment the device is "acquired" there is system event of device tree change, the application shall process it properly. If some previous version of the application works and current one does not, you can review changes that may cause changed behavior (assuming this is open source project).
You can do 'smoke test' of device access using UsbDkController.exe, which is under 'Programfiles/UsbDk...'. From administrator command prompt navigate to this directory, run "usbdkcontroller -n', see list of devices, then issue 'usbdkcontroller -r "$(device_id)" serial_number (exactly as they appear in "-n" output).

PipeRojas, 2020-08-31:
I can confirm this issue was solved in the newest versions of Dolphin Emulator. Im closing this issue

closed 2020-08-31 by PipeRojas

Squall-Leonhart, 2022-12-28:
It was not fixed, dolphin stopped using usbdk entirely pending an actual resolution.