daynix/UsbDk #61: USBDK report wrong USB speed under win7

crab2313, 2018-06-22:
I have issue with USBDK under windows 7 with the latest USBDK. It seems that USBDK will report a working super speed device as a high speed device. I have tried other backends of libusb (WinUSB)  and it reports correct usb speed. 

I have tested it on a laptop and QEMU(virt-manager with USB 3.0 host controller) with a host controller driver of version 2.1.39.0. I will post a trace log soon.

ybendito, 2018-10-14:
Which USB3 controller drivers you installed for USB3 controller? (If I'm not mistaken, Windows does not provide drivers for USB3 controller via Win7 Windows Update).

crab2313, 2018-10-16:
Driver is provided by renesas.

ybendito, 2018-10-18:
Fixed
https://github.com/daynix/UsbDk/commit/4d1ecea50dc99add7d0bc2ff4bb974403941c5e4

closed 2018-10-18 by ybendito