daynix/UsbDk #69: Installing UsbDk 1.0.19_x64 on windows 10 cause USB ports to die during Intallation or HotPlug attempts until next reboot

qubeck, 2019-02-12:
Hello,

Trying to install UsbDk 1.0.19_x64 on windows 10 causes USB ports to die occasionally during the installation but certainly after a subsequent reboot after several HotPlug attempts…

If attached the USB hardware remains functional during a reboot but the ports become unavailable again after detaching/re-attaching couple times. Not all devices behind a hub are becoming unavailable if a hub is being re-attached but eventually all of them after several attempts.

ybendito, 2019-02-18:
Thanks you for the report. In the log file we do not see anything suspicious, so can you please help us understand the flow that causing the problem? Do you replug specific devices or entire hub? What is the status of non-working devices in the Device Manager (yellow bang or works OK). There is no logs from the moment of the installation, does the problem starts from installation time? Note that just after UsbDk installation there is short period (usually several seconds) that already connected devices need to become functional.

qubeck, 2019-02-22:
I'm just re-plugging the entire hub to point out that the devices are not gone all at once...
In the mean time I made some progress to narrow down the issue and it seems that it is related to the powered hub that is built-in in my monitors.
Reattaching it seems to cause the entire USB stack to freeze. Detached devices are still shown present in device manager regardless of the port they are connected.
New devices are not getting enumerated anymore (no update in device manager).
Attached is the lsusb output for the hub under Linux.
Maybe you are able to figure out what could potentially cause this particular issue.
[lsusb_hub.txt](https://github.com/daynix/UsbDk/files/2894704/lsusb_hub.txt)