daynix/UsbDk #70: UsbDk_StartRedirect sometimes never returns ( hangs )

GitMiNo, 2019-03-03:
Using UsbDk 1.0.019_x64 from a 32 bit program on Windows 10. The UsbDk_StartRedirect sometimes doesn't return. This doesn't happen often. So most of the time there is no problem but sometimes at any moment the startRedirect call hangs ( no hardware change ). When this happens, after killing and restarting the program the problem is gone. Other times it will keep failing and a reboot is required. On my PC this happened once in 2-3 weeks but other people reported this more often. 

BTW: I also notices when installing UsbDk 1.0.19 on some PC's that the USB isn't working anymore ( keyboard mouse etc ). After a reboot the problem is gone. I believe this problem is also reported 
by someone else.

ckelsel, 2019-05-10:
Yes, sometimes keyboard and mouse aren't working anymore in Windows 10

GitMiNo, 2019-05-13:
So there are two problems, the second problem i mention is the keyboard and mouse not working after installing UsbDk sometimes. This is not a big issue, since it happens only on some PC's and only one time after install. After reboot the keyboard and mouse never hangs again.

The first problem is a bigger problem that the "UsbDk_StartRedirect" hangs sometimes with the devices we use, resulting in a hanged program. About 15+ people at the company use it.   

Is this a known problem and are you planning to look into this issue or 
is it possible to cancel the "UsbDk_StartRedirect" or give it a timeout ?

ybendito, 2019-06-07:
@GitMiNo UsbDk_StartRedirect by design has timeout of 60 seconds and always shall return. Typically it succeeds almost immediately. In rare cases when the device/hub fail to process 'port cycle' request properly and the device does not come back in reasonable time (60 sec), the call shall return anyway. We suggest to make record of the problematic scenario as described in https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt and provide additional info about USB devices that cause such problem, see also https://github.com/daynix/UsbDk/blob/master/Documentation/UsbInfo.txt

ybendito, 2020-03-17:
Please try 1.00-22, it solves some problems with redirection of specific devices. Please reopen if needed, point on exact devices and provide a record.

closed 2020-03-17 by ybendito

msbroadf, 2020-04-11:
I can reliably cause Usbdk_StartRedirect to lock.

The device is a USB WiBU-BOX/U software dongle

I will also try to pin down the exact issue. It seems like the Port Power Cycle fails and 

The log follows

[3]2FC0.21CC::04/11/2020-23:55:58.120 [UsbDk]CUsbDkFilterDevice::OnFileCreate
[3]2FC0.21CC::04/11/2020-23:55:58.120 [UsbDk]CUsbDkFilterDevice::OnFileCreate: completed
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]<lambda_9d5f7a4922b3a9cec393bc6320049d82>::operator() Starting relations array processing:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CDeviceRelations::Dump Array size: 5 (ptr: FFFFAD8CA02E6910)
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CDeviceRelations::ForEachIf<bool #0: FFFFAD8C9B421570
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CDeviceRelations::ForEachIf<bool #1: FFFFAD8C9B448B40
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CDeviceRelations::ForEachIf<bool #2: FFFFAD8C9B4488F0
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CDeviceRelations::ForEachIf<bool #3: FFFFAD8C9B31FDC0
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CDeviceRelations::ForEachIf<bool #4: FFFFAD8C9B31DDC0
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B421570 already registered:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B421570:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: USB\VID_088E&PID_5036
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: 3FA0812A_491E0364
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B448B40 already registered:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B448B40:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: USB\VID_088E&PID_5036
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: 5F8AA52B_4A29123C
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B4488F0 already registered:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B4488F0:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: USB\VID_05C8&PID_0801
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: 0x0001
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B31FDC0 already registered:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B31FDC0:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: USB\VID_05C8&PID_0800
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: 200901010001
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B31DDC0 already registered:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B31DDC0:
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: USB\VID_8087&PID_0A2A
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]CRegText::Dump ID: 8
[0]0004.1A18::04/11/2020-23:55:58.250 [UsbDk]<lambda_9d5f7a4922b3a9cec393bc6320049d82>::operator() Finished relations array processing
[1]100C.1BA8::04/11/2020-23:55:58.446 [UsbDk]CUsbDkFilterDevice::OnFileCreate
[1]100C.1BA8::04/11/2020-23:55:58.446 [UsbDk]CUsbDkFilterDevice::OnFileCreate: completed
[0]100C.1BA8::04/11/2020-23:55:58.507 [UsbDk]CUsbDkFilterDevice::OnFileCreate
[0]100C.1BA8::04/11/2020-23:55:58.507 [UsbDk]CUsbDkFilterDevice::OnFileCreate: completed
[2]2FC0.21CC::04/11/2020-23:55:58.983 [UsbDk]CUsbDkFilterDevice::OnFileCreate
[2]2FC0.21CC::04/11/2020-23:55:58.983 [UsbDk]CUsbDkFilterDevice::OnFileCreate: completed
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]<lambda_9d5f7a4922b3a9cec393bc6320049d82>::operator() Starting relations array processing:
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CDeviceRelations::Dump Array size: 5 (ptr: FFFFAD8CA0288A90)
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CDeviceRelations::ForEachIf<bool #0: FFFFAD8C9B421570
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CDeviceRelations::ForEachIf<bool #1: FFFFAD8C9B448B40
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CDeviceRelations::ForEachIf<bool #2: FFFFAD8C9B4488F0
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CDeviceRelations::ForEachIf<bool #3: FFFFAD8C9B31FDC0
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CDeviceRelations::ForEachIf<bool #4: FFFFAD8C9B31DDC0
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B421570 already registered:
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B421570:
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CRegText::Dump ID: USB\VID_088E&PID_5036
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CRegText::Dump ID: 3FA0812A_491E0364
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CWdmList<CUsbDkChildDevice,CWdmSpinLock,CCountingObject,CScalarDeleter<CUsbDkChildDevice> PDO FFFFAD8C9B448B40 already registered:
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CUsbDkChildDevice::Dump Child device 0xFFFFAD8C9B448B40:
[3]0004.1F6C::04/11/2020-23:55:59.092 [UsbDk]CRegText::Dump ID: USB\VID_088E&PID_5036
[... cut ...]
[2]100C.0D30::04/11/2020-23:58:34.033 [UsbDk]CUsbDkFilterDevice::OnFileCreate: completed
[2]100C.0D30::04/11/2020-23:58:34.035 [UsbDk]CUsbDkFilterDevice::OnFileCreate
[2]100C.0D30::04/11/2020-23:58:34.035 [UsbDk]CUsbDkFilterDevice::OnFileCreate: completed
[1]2FC0.0D6C::04/11/2020-23:58:34.053 [UsbDk]CUsbDkControlDevice::AddRedirect Wait for redirector attachment failed. 0x00000102(STATUS_TIMEOUT)
[1]2FC0.0D6C::04/11/2020-23:58:34.053 [UsbDk]CUsbDkRedirection::Dump Deleting  DevID: USB\VID_064F&PID_0BD7, InstanceID: 3

msbroadf, 2020-04-11:
Some more information:

If the wibubox that is redirected DOES NOT have a driver on the host (In device manager it shows an exclamation with no driver as the reason) then it will lock when start/stop/start redirect (sometimes just start redirect the first time will lock)

 If the device DOES have a driver bound to it on the host before it is redirected there is no lockup. If i bind a driver to the wibu-box/u then redirected it afterwards via usbdk it will not lock no matter how many times i start/stop redirection