daynix/UsbDk #72: Usbredirect HP-printer leads to Windows blue screen

wangjiedong, 2019-04-02:
I use my Windows 7 with virt-viewer-6.0 and UsbDk-1.0.19 as client,
then connect to a Ubuntu-16.04 guest , and i can select my U-disk to usbredirect successfully ,
but when i select the HP Color LaserJet M552 printer to redirect , the Windows blue screen occurs at once .
I can use this printer in Win7 normally , and after i check a 4G memory bank then redirect printer, blue screen again .

ybendito, 2019-04-15:
Can you please provide the system memory dump?
1. Configure the system to create kernel memory dump (Control Panel - System - Advanced system setting - Startup and Recovery - Settings - Select 'Kernel Memory dump' - Check 'Overwrite'.
2. After the crash happens and the system reboots - collect %windir%\memory.dmp, zip it and share with us using google drive / dropbox / any other file sharing service.
Thanks

ybendito, 2020-03-17:
1.00-22 released, should fix the problem. Please reopen if needed

closed 2020-03-17 by ybendito