daynix/UsbDk #75: usbdk windows xp  administrator rights

cobradevil, 2019-05-15:
I have a question regarding usb forwarding using a privileged user on xp.
I know xp is old but i have this use case so would like to see if it can work.

The thing is that when i install usbdk the driver gets installed but it does also want to install the drivers for the virtual device when a redirection occurs (2b23:cafe). This step can only be done with a user that is member of the administrator group. After that it works correctly with all usb devices. Is there a way todo this using a script or something without redirecting the device?

Best regards,
William

ybendito, 2019-05-27:
@cobradevil If I'm not mistaken, there is no requirement for administrator priviliges for redirection. The hidden device is 'raw' one and does not have the function driver. Can you please describe the exact flow and the problem (when it happens and when does not)?

cobradevil, 2019-05-27:
My setup is:
windows xp using usbdk 1.0.21

The user has limited rights (only user rights).
I installed usbdk 1.0.21 as admin on the machine using the msi.
I start remote-viewer 2.0-179 to a windows 7 session as the limited user.
After plugging in the usb mass storage device on the xp machine a xp popup box comes onscreen telling me that i do not have the proper rights to add the usbdk device to the system. When ignorig the popup the device still works as expected but the popup is very inconvenient.

I also tried using a hiding rule but the results are the same.

Best regards,
William van de Velde

cobradevil, 2019-05-27:
And if i use the hider rule should the device be completely unavailable for the host os?
so if i plugin the usb mass storage device with the above rule with no spice session running then the device should be unavailable on xp?

ybendito, 2019-05-27:
@cobradevil Hiding with -P causes the matching devices to be always hidden regardless spice session

cobradevil, 2019-05-27:
@ybendito so usbdk does create a local device even when it should not do it.

As a workaround i now added a registry setting suppressing the new hardware found wizard so no popup comes to the screen. But i thought the hiding rule would prevent that window from popping up.

ybendito, 2019-05-27:
@cobradevil UsbDk does not create the device, it is already created by the system. When the device is hidden or redirected, the UsbDk changes its attribute to be 'raw', so the device does not require driver. It also sets flag 'SilentInstall' for the device, so the OS should not raise any popups for it. Probably, XP ignores it. AFAIR, such problem does not exist in Win7 and up

cobradevil, 2019-05-27:
@ybendito 

I will test the registry setting. Maybe there could be a line in the readme which addresses this issue?

Best regards,
William van de Velde