daynix/UsbDk #78: uninstall usbdk doesn't remove the usb filters

boistordu, 2019-09-16:
Hi,

windows 10 1903 pro here.
When uninstalling usbdk runtime libraries 1.0.21 and also 1.0.19 it doesn't remove the usb filters so I stay stucked with it.
Could you provide a complete list of items that are installed with the windows installer which is the drivers, which peripherals in the device managers would be concerned and the registry keys?
Thanks in advance

ybendito, 2019-09-19:
According to our check, both 1.0.19 and 1.0.21 do remove the usb class filter installed by USBDK on uninstall. When installed, the UsbDk installer populates "UpperFilters" value under HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000} and reenumerates all the USB controllers. On uninstall, it removes this value and reenumerates USB controllers again. Please provide more information about the problem. Also, in order to install/uninstall just the filter driver (after UsbDk.msi is installed) you can use following sequence from administrator command line: cd to "%programfiles%\UsbDk Runtime Library", then usbdkcontroller -u (for uninstall) or -i (for install). UsbDk.msi populates files under "%programfiles%\UsbDk Runtime Library" and UsbDkHelper.dll under system32 and syswow64 (if applicable)

ybendito, 2019-11-23:
Closing the issue. Please reopen if still actual.

closed 2019-11-23 by ybendito

ybendito, 2019-11-23:
Closing the issue, please reopen if still actual