daynix/UsbDk #79: usbdk1.0.19 update to 1.0.21, windows blue screen

LAIhy, 2019-11-21:
I use UsbDk 1.0.19 in my pc, windows 7, It work fine, but when I update usbdk to 1.0.21, I run the UsbDk_1.0.21_x64.msi in my pc, the windows show blue screen. And after I reboot my pc, I uninstall the usbdk 1.0.19, install the usbdk 1.0.21 again, this problem still exist. Is there any way to fix this problem？
[image]

ybendito, 2019-11-23:
Question 1: If you uninstall 1.0.19, then install 1.0.19 - does the problem happen or not?
In any case I recommend to uninstall 1.0.19, use Windows Update to take latest hot fixes, then try again to install 1.0.21. If the problem persists, we'll try to find a root cause together.

LAIhy, 2019-11-28:
Question: If you uninstall 1.0.19, then install 1.0.19 - does the problem happen or not?
=>yes，this problem still happen

It seem that the  original USB driver of my pc has been damaged，and I finally fix this problem in the following steps
1、delete usbehci.sys, usbhub.sys, usbohci.sys, usbport.sys in C:\\windows\system32\drives
2、uninstall all USB drivers
3、delete  usbui.dll in C:\\windows\system32
4、reboot the computer

after reboot, the problem solved, and I can install usbdk 1.0.19 or usbdk 1.0.21 successfully

closed 2020-01-02 by ybendito