daynix/UsbDk #81: Problems with USBDK & GIMX

JusticePug, 2019-12-27:
Hi everybody,
I was told the best way to solve my problem is to open an issue regarding GIMX and USBDK.
For GIMX itself I need USBDK to use my PS4 controller on my PC, I've tried different versions, using GIMX without USBDK and whatelse you can think of. Hopefully one of you guys can help me.

As soon as i start up GIMX I get a BSoD instantly, analyzing the dump files "usbdk.sys" is the problem regarding the BS. I've tried it with v1.0.19 and v1.0.21. I had them removed, rebooted PC, reinstalled USBDK, rebooted again. Nothing works.
I'll attach the .dmp files.
Maybe you can find the problems.
As a system i use an AMD build with Win10 1909 64bit if that helps.

[Dumpfiles_USBDK.zip](https://github.com/daynix/UsbDk/files/4005823/Dumpfiles_USBDK.zip)

matlo, 2019-12-27:
In addition, latest GIMX version (7.15) uses libusb 1.0.21 with the usbdk backend. I have no issue on my own PC which runs Windows 10 1909 (18363.535).

ybendito, 2019-12-29:
@JusticePug Provided dump files are minidumps and do not contain enough info to diagnose the problem. I suggest following (with UsbDk 1.0.21):
1. Configure the system for kernel crash dump
2. Use https://github.com/daynix/UsbDk/blob/master/Tools/Trace/UsbDk-Autologger-On.reg to activate automatic recording. Reboot.
3. Reproduce the problem, %windir%\memory.dmp will contain the dump of physical memory. I think you have at least 16G of memory, so it will be large. After the problem happens, zip the dmp fie and put on google drive / dropbox. We'll look at it. Upload also system32\logfiles\wmi\usbdk.etl file (collected after crash).
4. When you want to turn off the automatic recording, use https://github.com/daynix/UsbDk/blob/master/Tools/Trace/UsbDk-Autologger-Off.reg
Do you have 32-bit or 64-bit installation of the GIMX (whatever it is)?
Did it work previously on your machine?
@matlo 
Is it possible to use the GIMX without UsbDk?

matlo, 2019-12-29:
Installed GIMX is version 7.15 x86_64, which redistributes UsbDk_1.0.19_x64.msi.

@ybendito 
Not sure what you mean by "without UsbDk", but GIMX does not use UsbDk on some use cases. I did not try any alternative libusb backend lately. The other backends sadly lack features that the UsbDk one has.

JusticePug, 2019-12-29:
> 
> 
> Installed GIMX is version 7.15 x86_64, which redistributes UsbDk_1.0.19_x64.msi.
> 
> @ybendito
> Not sure what you mean by "without UsbDk", but GIMX does not use UsbDk on some use cases. I did not try any alternative libusb backend lately. The other backends sadly lack features that the UsbDk one has.

It's not possible to start GIMX without USBDK installed.
It wont find the DS4 controller in that case Matlo. I've tried without it. :)

JusticePug, 2019-12-29:
@ybendito Just to make sure.
The given path gets me to code, problem now is, I don't have a UsbDk Folder in my registry and I can't add one from what I've seen (no big fan of messing around in registry myself as it's super sketchy after all imo)
What is my next step to do? I feel like I'm missing something...

matlo, 2019-12-29:
@JusticePug 
Actually for the PS3 use case does not require UsbDk.

JusticePug, 2019-12-29:
Good to know.
Sadly I'd like to use the adapter with PS4 so I don't think a PS3 Controller does the job.

matlo, 2019-12-29:
@JusticePug 
Indeed. I was just trying to answer @ybendito's questions ;)

matlo, 2019-12-29:
@JusticePug 
The easiest way to proceed is to save the registry files linked above and to double click them to have them applied. The registry keys added here only apply to UsbDk, so they are safe to add.

ybendito, 2019-12-29:
@JusticePug This is not a code, this is a .reg file, all you need is just double-click on it and it will be inserted to the registry.

JusticePug, 2019-12-29:
I'm fkin bad at code, literally no idea (my grades confirm that) & it's the first time I'm really using Github so now it took me a while to find how to download it... didn't know it was the full folder to get :D
Will do the above steps and get you the needed files.

JusticePug, 2019-12-29:
Alright got it done.
https://drive.google.com/drive/folders/1YUuskxejDKvCZMEwssLlJB5amrSFL9ZH?usp=sharing
I hope the link works as intended.
For clarification, it's the 64bit version of GIMX I'm using. 32bit won't run for the same reason as with 64bit.

ybendito, 2019-12-30:
@JusticePug 
Assuming you do not run with secure boot enabled. To ensure, run msinfo32 and look for "Secure boot state". If it is not like "On" or "Enabled", continue.
1. Install usbdk 1.0.21 (note that the dump was with 1.0.19)
2. Enable usbdk logging as above (if disabled)
2. Enable test signature: From admin command line: "bcdedit /set testsigning on", reboot
3. Unzip attached usbdk.sys and place it in "%program files%\usbdk runtime library" over existing one
4. From admin command line:
cd c:\program files\usbdk runtime library
usbdkcontroller -u
usbdkcontroller -i
Execution of both commands takes some time.
5. If everything looks ok, try the gimx
6. If something bad happens, upload the dump as before. If not, upload the usbdk.etl
[UsbDk.zip](https://github.com/daynix/UsbDk/files/4009862/UsbDk.zip)

JusticePug, 2019-12-31:
Got to step 5 and now command line shows me that the installation failed after uninstalling usbdk driver...
`c:\Programme\UsbDk Runtime Library>usbdkcontroller -i
Installing UsbDk driver
UsbDk driver installation failed`

ybendito, 2020-01-01:
@JusticePug Probably lack of reboot after step 3?

JusticePug, 2020-01-01:
Did the reboot after enabling test signature.
Even rebooted afterwards again, installation still failing.

ybendito, 2020-01-01:
Try to delete c:\windows\system32\drivers\usbdk.sys from admin command line and retry the "usbdkcontroller -i"

JusticePug, 2020-01-01:
Driver installation still failing after deleting usbdk.sys in c:\windows\system32\drivers\

ybendito, 2020-01-01:
Download
https://docs.microsoft.com/en-us/sysinternals/downloads/debugview
Runs it as an administrator, menu -> "capture" -> enabled capture Win32  + capture global Win32.
Retry "usbdkcontroller -i", in the debugview should be a printout of the failure reason

JusticePug, 2020-01-01:
Seems like it got throu without any problems as soon as i had debugview started.
The follwing part is the next issue I ran into. I dont know if this is an only GIMX related problem or again the combination of both. I had this error before aswell...
`src/gusb.c:865 claim_device: libusb_open failed with error: Other error
Error: No game controller was found on USB ports.`

ybendito, 2020-01-01:
@JusticePug The blue screen does not happen with new usbdk.sys? Please upload usbdk.etl
" I had this error before as well" - how is it, if previously it was blue screen on startup of the application?
Look at device manager: do you have some usb devices without drivers installed?

JusticePug, 2020-01-01:
So the thing with Usbdk seems to be fixed since it's not bluescreening anymore.
Now there is just an issue with the DS4 controller not being detected.
The error that occured now is the same i had with usbdk not being installed on the PC so I'll get back after I fiddled around some more :)

matlo, 2020-01-01:
> src/gusb.c:865 claim_device: libusb_open failed with error: Other error
> Error: No game controller was found on USB ports.

This error comes from libusb_open: https://github.com/libusb/libusb/blob/v1.0.21/libusb/os/windows_usbdk.c#L424
I think the underlying UsbDk call is UsbDk_StartRedirect.

JusticePug, 2020-01-01:
Alright.
Funny how i just connect it to my new 2nd PC and GIMX is instantly working and i can use my Wheel on my PS4.
For me it works on my Intel built PC but not on the Ryzen one. For my frind it's the exact opposite with GIMX working on Ryzen but not on Intel.
If you wanna keep up investigating the error i'll be glad to help out but for me the issue is resolved atleast as a workaround :)

ybendito, 2020-01-01:
@JusticePug Do you have lghub.exe and lghub_agent.exe processes running on both machines? (task manager)

matlo, 2020-01-01:
I installed lghub and reproduced the BSoD with UsbDk 1.0.21. After replacing usbdk.sys I am not getting the BSoD, but GIMX cannot open the Dualshock 4 device while lghub is running. After terminating lghub and lghub_agent GIMX can open the Dualshock 4 device.

ybendito, 2020-01-01:
@matlo Thanks, this is what I've suspected. There is some conflict of what lghub does (nothing criminal) with what UsbDk expects to see.

matlo, 2020-01-01:
@ybendito Is there any way to fix this conflict on the UsbDk side or does the fix need changes on the Logitech side?

ybendito, 2020-01-01:
@matlo In general, I think, yes. The devil is in details. Can you explain: what the lghub is good for? Do I understand correctly that lghub is a service? Does it add some value to gimx or they are orthogonal? For example, does stopping lghub when gimx starts provide the reasonable workaround?

matlo, 2020-01-01:
@ybendito I'm a bit confused. Should we expect a fix on the UsbDk side?

In case it helps here's a new log with latest failure: [UsbDk.zip](https://github.com/daynix/UsbDk/files/4014446/UsbDk.zip)

ybendito, 2020-01-01:
@matlo Blue screen will be definitely fixed in the next release. We'll discuss the issue with lghub in our meeting.

ybendito, 2020-01-02:
Please check whether attached driver solves the problem.
Installation instructions as above https://github.com/daynix/UsbDk/issues/81#issuecomment-569676640
Please provide the driver log whether it solves the problem or not.
[UsbDk.zip](https://github.com/daynix/UsbDk/files/4015975/UsbDk.zip)

matlo, 2020-01-02:
It works, thanks!

I did the following steps:
* replace UsbDk.sys
* usbdkcontroller -u
* enable log
* usbdkcontroller -i # failed since I had test mode disabled
* turn test mode on and reboot
* usbdkcontroller -u
* usbdkcontroller -i # still failed, no idea why
* remove completely UsbDk then reinstall 1.0.21
* replace UsbDk.sys (re-downloaded to be sure it is the right one)
* usbdkcontroller -u
* usbdkcontroller -i # succeed
* use GIMX # succeed

ybendito, 2020-01-02:
The log seems ok. I leave the issue open, will close it when the fix is released.

matlo, 2020-01-02:
If you have any further tests feel free to ask!

matlo, 2020-01-02:
I forgot to ask, do you have any estimate for the release date?

ybendito, 2020-03-17:
1.00-22 released, should fix the problem. Please reopen the issue if needed

closed 2020-03-17 by ybendito