daynix/UsbDk #92: Windows on ARM support?

Lonewolf81, 2020-07-21:
Hi

Windows on arm has the ability to support x86 programs but I have tried to install this even with driver signature enforcement off. Will this version work on it or does it have to be compiled for arm64 and if it is compiled will it still have support for x86 emulation ?

YanVugenfirer, 2020-07-21:
Hi,

In Windows OS you must install the driver compiled for the specific platform (even on x86 you have 32 bit and 64bit drivers).

Lonewolf81, 2020-07-21:
Hi Daynix/Usbdk

Thank you for getting back to me so quickly, so basically it would have to
be compiled for the Windows on Arm version I am testing on. I have no
experience with this but will have a look if I can find the correct method.

Regards
Friedrich Rabie

YanVugenfirer, 2020-07-22:
Hi,

For now, we don't have the option to compile USBDk for ARM64.
This is something we need to put on a roadmap. If more people are interested, please reply on this issue.

Thanks,
Yan.

sidd-kishan, 2021-05-29:
Yes Please pursue this.

sidd-kishan, 2021-05-30:
I would like to sponsor this compilation ARM64 windows on arm driver.

YanVugenfirer, 2021-05-30:
@sidd-kishan you can contact me for more details: yan@daynix.com

CharlesForson, 2021-08-28:
Hello! Also interested if it's not too late. No x86_64 machine at the moment, so working with the humble rpi4b to revive a gm1917. Sorry if this is no longer the right place to inquire (your email is linked in the previous message, but the message was a reply to another user). Thanks!

ps...noticed the pic. You a Sturgis guy?

CharlesForson, 2021-08-28:
Oh, not sure if it's relevant, but found your repo via bkerler's edl tool. Thanks again!

YanVugenfirer, 2021-08-29:
@CharlesForson I am riding Harley, but on another continent. Hopefully, I can visit Sturgis one day :)

The ARM port for USBDK is in the pipeline, but it is hard to say when we will start working on it. So, unfortunately, there are no binaries for ARM.

george-howell, 2021-12-28:
Yes, I would like this as well. It seems to be a part of the ADI CCES installation, and it is failing when trying to start the driver. 

I've posted a question on ADI's forum, not much hope that they will get back though, let alone promptly.

george-howell, 2022-01-05:
Fair play to ADI, they did get back to me quite quickly and said they'd look into it. Whether or not they'd be willing to be a sponsor though I don't know.

ormembar, 2023-08-06:
I see the last post has been a while
Is there any news on this? I am now on an ARM64 device and would have use for the driver.

hjalmarnilsson, 2024-03-15:
Nudging on this, I may also want to contribute (money wise) to push this forward. Whats the latest greatest on this subject?

YanVugenfirer, 2024-03-17:
Let's discuss it over email yan @ daynix.com

Daplugg23, 2024-05-27:
Please add Arm support

harojpgyy, 2024-06-06:
When will arm be supported?

YanVugenfirer, 2024-06-06:
Unfortunately we don't have resources to invest in it now.
If you are a commercial organisation, please contact me directly.