daynix/UsbDk #97: BSoD when keyboard smart card was redirected on Windows 10 and virt-viewer

FutureKidBuu, 2020-11-16:
We have a rule for virt-viewer to redirect on connect all usb devices and a BSoD image always happens to this smartcard keyboard https://unykach.com/es/perifericos/teclado/teclado-sck-818-inteligente-50530/ when we connect a virtual machine.

It was tested with other smart card keyboards and it worked fine.

**Important data:**

- **Memory dump file:** https://tinyurl.com/y5sqhlj4

- **Smartcard keyboard product:** https://unykach.com/es/perifericos/teclado/teclado-sck-818-inteligente-50530/

- **Windows 10 version:** 2h20

- **BSoD:** [image]

- **If I uninstall the usbdk package:**

[image]

ybendito, 2020-11-16:
@FutureKidBuu  We can't access the memory dump link, can you please make it available?
Which version of UsbDk you use?

FutureKidBuu, 2020-11-16:
@ybendito UsbDk is the last one, v1.00-22.
Please, test link again, https://tinyurl.com/y5sqhlj4

ybendito, 2020-11-16:
@FutureKidBuu Can't access, tried with 2 different Microsoft  accounts (it requires sign-in with Microsoft). Tyhe test as always the same: We're sorry, but <mail address> can't be found in the activacionteknoservice-my.sharepoint.com directory. Please try again later, while we try to automatically fix this for you

FutureKidBuu, 2020-11-16:
@ybendito sorry for that,I upload it to my drive https://drive.google.com/uc?id=15RVxvR5wKQEOjo43-JKnFdyK6-fqGx2K&export=download
I hope you can!

ybendito, 2020-11-18:
@FutureKidBuu The dump downloaded. Next time you share large binary files, please zip them. For size as well as to prevent a corruption.
Please try previous release https://www.spice-space.org/download/windows/usbdk/UsbDk_1.0.21_x64.msi
(uninstall 1.0.22 first, then install it). Does it solve the problem?

FutureKidBuu, 2020-11-19:
@ybendito with version 1.0.21 it works! should we use that version as the last stable?

ybendito, 2020-11-19:
Latest stable is 1.0.22 but it causes the problem with this specific device USB\VID_058F&PID_9540 (EMV Smartcard Reader)
If you use 1.0.22 you need to exclude this device (smartcard reader) from the redirection (you can do that via configuration file of remote viewer or via its command line or on spice server side, i.e. in qemu command line (if you use qemu)
I think if you disable power save on this device (uncheck 'allow the computer to turn off this device ...') this will also solve the problem.

celevra, 2021-08-30:
same problem with the bluescreen here, but we need the Smartcard in the virtual machine forwarded by remote viewer
i can't tell with update, but before a specific Windows Update forwarding went smoothly

ybendito, 2021-09-14:
@celevra At the moment I see 2 things to try:
- Use 1.0.21 if possible
- Disable USB selective suspend (Control Panel - Power - Current power plan)